carla-simulator/carla #3260: Research work on maps

xiaoyang666, 2020-09-03:
Research work on maps
Since roadrunner is not free, I have launched some attempts here
1. I bought a map in the UE4 store, generated a blueprint class from the c++roadrunner class, and manually added the car spawning point and roadplanner to the map, but this is currently unavailable
2. Use the osm file to input into the sumo simulator to obtain the opendrive file, generate the fbx file from the osm file according to cityengine, and import it into carla, there are still errors

As far as I know, in the early version of carla such as 0.8.0, it seems that the roadrunner software is not used, so I came up with the idea of 1, which is still impossible to achieve at present.

Will the latest 0.9.10 solve problems similar to 2? I look forward to it

germanros1987, 2020-09-24:
@Axel1092 could you provide some guidance about map creation?

Axel1092, 2020-09-24:
Hi @xiaoyang666, sumo's conversion tool to OpenDRIVE format produces broken OpenDRIVE files that cause Carla to crash so this is probably the biggest issue you are facing. In 0.9.10 we provide a osm to OpenDRIVE conversion tool without this problem, you may try that feature. Anyways, could you provide us with the errors you are encountering and their console outputs in order to take a better look at your problem?

closed 2020-12-25 by stale[bot]