carla-simulator/carla #3269: opendrive could not be correctly parsed

HemaZ, 2020-09-07:
Hello CARLA Team, 
I'm having an issue running the opendrive standalone example in the documentation 
`python3 config.py -x opendrive/TownBig.xodr
`
I get the following error 
_**opendrive could not be correctly parsed**_
I'm using CARLA Patch Release 0.9.9.4
on Ubuntu 18.04 inside Docker
I've tried using python3.6 and python3.7 but I get the same issue. 
Thank You.

germanros1987, 2020-09-24:
@HemaZ could you please check if you have the same problem in the nightly build? http://carla-releases.s3.amazonaws.com/Linux/Dev/CARLA_Latest.tar.gz

HemaZ, 2020-09-24:
> @HemaZ could you please check if you have the same problem in the nightly build? http://carla-releases.s3.amazonaws.com/Linux/Dev/CARLA_Latest.tar.gz

Nightly build is working great 
[image]

HemaZ, 2020-09-25:
@germanros1987 I've tried again using the nightly build and it returns the same error !!

closed 2020-12-25 by stale[bot]

InterKnight, 2021-01-18:
I got the same error. Anyone can give a hint?
I'm using CARLA 0.9.11 from deb package on Ubuntu 18.04

egreglee-bit, 2021-02-01:
Similar failure as in other reports, but from CARLA_0.9.11.tar.gz on Ubuntu 18.04
I verified with an independent viewer that the xodr can be parsed.

bitekas, 2021-02-23:
I'm confirming the same bug, ubuntu 18.04

HarnarayanSingh, 2021-02-23:
> I'm confirming the same bug, ubuntu 18.04; CARLA_0.9.11; the nightly build by the link is working

@bitekas Could you clarify if it worked or not for you? I got the same error in nightly build.

bitekas, 2021-03-01:
@HarnarayanSingh Yes, the nightly build is working for me.

LichuanRen, 2021-03-27:
I got the same error on Ubuntu 18.04, CARLA 0.9.10. 

The nightly build returns another error: RuntimeError: rpc::rpc_error during call in function load_new_episode