carla-simulator/carla #3270: Running CARLA from Docker failed

Leon-Adler, 2020-09-08:
Hello community!

After we installed,
- Docker and 
- nvidia-docker-2

we followed the instructions https://carla.readthedocs.io/en/latest/build_docker/.

In brief, we pulled the CARLA image 0.9.9
```
docker pull carlasim/carla:0.9.9
```

and ran CARLA under docker.
```
docker run -p 2000-2002:2000-2002 --runtime=nvidia --gpus all carlasim/carla:0.9.9
```

Unfortunately after running the command above we got the following error message:

```
sh: 1: xdg-user-dir: not found
ALSA lib confmisc.c:768:(parse_card) cannot find card '0'
ALSA lib conf.c:4292:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4292:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1251:(snd_func_refer) error evaluating name
ALSA lib conf.c:4292:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4771:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM default
ALSA lib confmisc.c:768:(parse_card) cannot find card '0'
ALSA lib conf.c:4292:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4292:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1251:(snd_func_refer) error evaluating name
ALSA lib conf.c:4292:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4771:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM default
```

Has anyone an idea to solve this?

Thanks for your contributions!

germanros1987, 2020-09-24:
@Leon-Adler it didn't fail. It just has an annoying problem with ALSA lib, but there is no segfault. Did you try to connect to the server?

joel-mb, 2020-09-25:
Hi @Leon-Adler !
As @germanros1987 said, these errors are not affecting the server. You should be able to run your client without problems.

kargarisaac, 2020-10-08:
I try to run Carla using docker on a remote server. I use the command `docker run -it -p 2002-2004:2002-2004 -e NVIDIA_VISIBLE_DEVICES=1 carlasim/carla:0.9.6` but get the following error:

```
4.22.1-0+++UE4+Release-4.22 517 0
Disabling core dumps.
sh: 1: xdg-user-dir: not found
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=48848 LargeMemoryPoolOffset=179984 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
CarlaUE4.sh: line 5:    13 Segmentation fault      (core dumped) "$UE4_PROJECT_ROOT/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping" CarlaUE4 $@
```

I want to run headless carla. There is one Carla running on another user account on this PC outside of docker. Does it affect? 

I also get the following error by running `docker run -p 2000-2002:2000-2002 --runtime=nvidia --gpus all carlasim/carla:0.9.6`:
```
docker: Error response from daemon: Unknown runtime specified nvidia.
See 'docker run --help'.
```

The tutorial on the website is for v8.2. Isn't it old?

pprivulet, 2020-11-06:
sudo docker run --gpus 0 --env SDL_VIDEODRIVER=offscreen carlasim/carla:0.9.10 /bin/bash CarlaUE4.sh
it works for me
good luck

closed 2021-07-21 by stale[bot]