carla-simulator/carla #3273: Import traffic signals but don't work.

wangludong, 2020-09-09:
Hi there

I made a map with traffic signals in RoadRunner.
When I imported the Opendrive and started the simulation, the traffic signals did't work.
The lights neither turned on nor switched.
I wonder what is the mechanism of traffic signals.Are they controlled by Opendrive or some other way?

thytless, 2020-09-09:
https://github.com/carla-simulator/carla/releases
Traffic lights auto-importing is supported since 0.9.9.1, please make sure you are using a higher version

wangludong, 2020-09-10:
> https://github.com/carla-simulator/carla/releases
> Traffic lights auto-importing is supported since 0.9.9.1, please make sure you are using a higher version

Thank you.
But I have to use 0.9.8 because of the limit of my project.
Is there any way to work with traffic lights in this version?

thytless, 2020-09-10:
Well, as far as I know you can always manually add traffic lights actors to your map in CarlaUE4 Editor.

wangludong, 2020-09-11:
> Well, as far as I know you can always manually add traffic lights actors to your map in CarlaUE4 Editor.

Thank you.
I added some BP_TrafficLights to my scene and they can switch lights.
But the lights of different directions turned to the same light simultaneously, as you can see in the following capture.
[image]

Do you know how to control them?

wangludong, 2020-09-11:
I associated a TrafficLightGroup with these TrafficLights and it became better.
But these lights circulated one by one instead of synchronizing the opposite lights so as to turn on the same light.
Need I modify the blue print?Or any other way exists?

germanros1987, 2020-09-24:
@wangludong you probably need to modify the BP, or you can update to 0.9.9...

Axel1092, 2020-09-24:
Hi @wangludong, at the version you are working you would need to do more than modify the TrafficLightGroup Blueprint and even the C++ part of the traffic lights control. The only workaround I can think of is to create a new traffic light Blueprint with 2 traffic lights in it and then adjust their position in the world manually.

In the latest version in our current master this feature is still not supported but it is a lot easier to workarround by only modifying the TrafficLightGroup BP.

Another solution in 0.9.9 and up is to remove the traffic lights from the map and let it load from the OpenDRIVE at runtime. You can place a `TrafficLightManager` actor in the world and provided they are well defined from roadrunner they will load and behave as you expect.

wangludong, 2020-09-30:
Okey, I get it. Thank you all!

closed 2020-09-30 by wangludong