carla-simulator/carla #3276: make launch doesn't build boost for python 2.7

abhi5691, 2020-09-09:
Using **Carla 0.9.9.4** on **Ubuntu 18.04** 

I am sure that both python 2.7 and 3.6 are functional on my system. However, I am only able to locate boost library files for python 3.6 but not for 2.7 inside my carla Build directory after a `make launch`. FYI, I have libboost-python36.a as well as libboost-python27.a in my usr/lib/x86_64-linux-gnu directory.
However since the carla Build is devoid of the boost lib for 2.7, I am unable to generate the corresponding egg file using `make PythonAPI`

Did someone face similar issues? Or has anyone got an idea what might be behind this?

P.S. Even if I try copying the 3.6 files and storing their copy as a corresponding 2.7 library file, the `make PythonAPI` process fails with error

>  ld status returned 1

Lu7913, 2020-09-16:
Similar thing happened to me when building Carla 0.9.10 on Ubuntu 18.04. With Carla 0.9.9.4 both egg files were generated (python 2.7 and 3.6), but with the 0.9.10, only egg file generated is the one for python 3.6.

abhi5691, 2020-09-16:
@Lu7913 I think in v0.9.10, you can generate egg file for Python2 by explicitly using command `make PythonAPI.2`

This is ofcourse after you have libboost for Py2
Hope this helps :)

abhi5691, 2020-09-16:
FYI: I installed libboost for python2 on my system, and upgraded to the latest Carla source (0.9.10). I found relevant libboost files for both Python 2 & 3 after a `make launch` call. Going forward, `make PythonAPI` call generates egg file only for Python3. To explicitly generate Python 2 egg file I used `make PythonAPI.2` as I mentioned above

closed 2020-09-16 by abhi5691