carla-simulator/carla #3278: Missing pyplusplus from Bitbucket URL (RSS Build docs)

kino3, 2020-09-10:
I'm trying to build RSS environment on my Ubuntu.
In https://carla.readthedocs.io/en/latest/adv_rss/ 
```
sudo apt-get install castxml
pip install --user pygccxml
pip install --user https://bitbucket.org/ompl/pyplusplus/get/1.8.1.zip
```
But we cannot get pyplusplus from this URL. Can we use the following one instead?
```
pip install --user pyplusplus==1.8.1
```

germanros1987, 2020-09-24:
@fpasch could you have a look?

berndgassmann, 2020-09-24:
See the latest docu here:
https://ad-map-access.readthedocs.io/en/latest/BUILDING/index.html
https://intel.github.io/ad-rss-lib/BUILDING/

just install via pip now ;-)
Will update the docu inside CARLA to fix this.

PR #3357 (2020-09-24): Upddate adv_rss docu

#### Description

Updated adv_rss.md build instruction to reflect:
- latest pyplusplus changes
- PROJ not being installed in the system anymore
- move to python3 on CARLA build.

Fixes #3278   

#### Where has this been tested?

  * **Platform(s):** ...
  * **Python version(s):** ...
  * **Unreal Engine version(s):** ...

#### Possible Drawbacks

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/3357)
closed 2020-09-24 by sergi-e