carla-simulator/carla #3279: Python API for appearance changes of the actors in the scene.

pavanteja295, 2020-09-10:
Hello,
I'm trying to recreate a single snapshot within CARLA by varying the appearances of the actors(vehicles). I can see from the Python API documentation (https://carla.readthedocs.io/en/latest/bp_library/) that one can select only the color and model of the vehicle but cannot choose the material of the vehicle. 

However, in the paper (http://proceedings.mlr.press/v78/dosovitskiy17a/dosovitskiy17a.pdf) the authors specifically mentioned that materials of a specific model are randomly sampled from a specified list of options. Can anyone help me our letting me know if deterministic material selection of a model vehicle is possible?

Thanks and Regards,
Pavan Teja Varigonda.

germanros1987, 2020-09-24:
@Madecu could you please help with this question?

Madecu, 2020-09-29:
Hi @pavanteja295, right now we have only exposed the color of the vehicles. If you want to change or add some particular color for each car, you can edit the "VehicleFactory" in the "Blueprints" folder and edit the colors.

[image]

pavanteja295, 2020-10-10:
Thanks a lot for the answer! What about the material customization rather than the color?

Madecu, 2020-10-13:
We haven't exposed any other parameter of the material yet. Although you can manually edit the material in unreal if this is what you want. You have to go to the car you want to edit, select the material slot where the paint job is applied and edit the material instance. We have some attributes to customize the material. 
I hope that helps

[image]

pavanteja295, 2020-10-15:
Thanks for the information. Just a last question is there any documentation to expand the default PythonAPIs to include some customized  additional APIs. For example for something as the above usecase.

sergi-e, 2020-10-15:
@pavanteja295 There is no specific documentation regarding how to expand the API, because that subject is extremely large and complex, depending on how and what do you want to do. 

It would be more a matter of how much programming do you know. I suppose that you are specifically interested in exposing those values in the API. A good option would be to take a look at the code, learn how similar blueprint attributes are exposed, an mimicking it for your interests.

closed 2021-04-27 by stale[bot]