carla-simulator/carla #3280: 0.9.10: LiDAR point clouds not displaying cars

SchDevel, 2020-09-10:
Hello together,

I am currently experiencing a problem regarding the LiDAR point clouds with CARLA, running on Ubuntu 18.04.
As shown in the attached screenshot, cars (red bounding box) are not reflecting any points. However, pedestrians work fine. This can be seen at the blue bounding box slightly behind the green one. The model / car with the LiDAR attached on has the green bounding box.

Is this some configuration problem?

Looking forward hearing from you.

Best Regards

SchDevel

[image]

marcgpuig, 2020-09-17:
Hi @SchDevel 
This is not a configurable parameter and it should work fine.
- Are you using the packaged version?
- In case you don't: did you modify the editor with custom or new cars?
- Do you know if it's a signle car model that is causing this? Which model it is?

I'm not able to reproduce your error on our current `dev` branch:

[image]

So if this was an issue in your current version, it will be fixed in 0.9.10.

Cheers!

germanros1987, 2020-09-24:
@SchDevel could you give us an update? Is this feature still not working for you?

SchDevel, 2020-09-24:
Hey @germanros1987 and @marcgpuig ,

sorry for my late response.
I actually did a clean install of the CARLA server to make sure that I did not mess up anything in the editor.

Indeed, when importing my own blueprint car, I somehow messed up and replaced other car-files, too.

The problem is now solved.
Thank you so much!

Best regards

SchDevel

closed 2020-09-24 by SchDevel

GarethZhang, 2021-03-30:
Hi @SchDevel ,
I imported my own blueprint car into Carla as you did. For some reason my LiDAR and semantic LiDAR sensor are not able to detect it in their scans. How were you able to solve it?

Thanks!

ParthAD3, 2021-04-14:
Did you get the reason @GarethZhang

GarethZhang, 2021-04-14:
Hi @ParthAD3 ,
I noticed that this is happening because I did not define custom collision in blueprint class in UE4. You can refer to section 4 on this page: [https://carla.readthedocs.io/en/latest/tuto_D_generate_colliders/#4-add-the-mesh-as-collider](url)

Hope this help!