carla-simulator/carla #3283: A question about  the explanation of  World.wait_for_tick() method in  official document

huan0808, 2020-09-11:
Hi,
    I think this function is a asynchronous method to make client wait server to compute a frame.And if it hasn't done,we can make a loop in client to wait the server.
[image]
    But in document's explanation it seems like a Synchronize method.
[image]
    This confuzed me and if my understand is wrong please tell me why.
Thanks!

germanros1987, 2020-09-24:
@sergi-e could you clarify this?

sergi-e, 2020-09-24:
Hi, @huan0808!

You are correct, the Python API docs are misleading. What do you think of this changes? 

- <a name="carla.World.on_tick"></a>**<font color="#7fb800">on_tick</font>**(<font color="#00a6ed">**self**</font>, <font color="#00a6ed">**callback**</font>)  
This method is used in [__asynchronous__ mode](https://[carla.readthedocs.io](#carla.readthedocs.io)/en/latest/adv_synchrony_timestep/). It starts callbacks from the client for the function defined as `callback`, and returns the ID of the callback. The function will be called everytime the server ticks. It requires a [carla.WorldSnapshot](#carla.WorldSnapshot) as argument, which can be retrieved from __<font color="#7fb800">wait_for_tick()</font>__. Use __<font color="#7fb800">remove_on_tick()</font>__ to stop the callbacks.  
    - **Parameters:**
        - `callback` (_[carla.WorldSnapshot](#carla.WorldSnapshot)_) – Function with a snapshot as compulsory parameter that will be called when the client receives a tick.  
    - **Return:** _int_  
- <a name="carla.World.remove_on_tick"></a>**<font color="#7fb800">remove_on_tick</font>**(<font color="#00a6ed">**self**</font>, <font color="#00a6ed">**callback_id**</font>)  
Stops the callback for `callback_id` started with __<font color="#7fb800">on_tick()</font>__.  
    - **Parameters:**
        - `callback_id` (_callback_) – The callback to be removed. The ID is returned when creating the callback.  
- <a name="carla.World.tick"></a>**<font color="#7fb800">tick</font>**(<font color="#00a6ed">**self**</font>, <font color="#00a6ed">**seconds**=10.0</font>)  
This method is used in [__synchronous__ mode](https://[carla.readthedocs.io](#carla.readthedocs.io)/en/latest/adv_synchrony_timestep/), when the server waits for a client tick before computing the next frame. This method will send the tick, and give way to the server. It returns the ID of the new frame computed by the server.  
    - **Parameters:**
        - `seconds` (_float_) – Maximum time in seconds the server should wait for a tick. It is set to `10.0` by default.  
    - **Return:** _int_  
    - **Note:** <font color="#8E8E8E">_If no tick is received in synchronous mode, the simulation will freeze. Also, if many ticks are received from different clients, there may be synchronization issues. Please read the docs about [synchronous mode](https://[carla.readthedocs.io](#carla.readthedocs.io)/en/latest/adv_synchrony_timestep/) to learn more.  
_</font>  
- <a name="carla.World.wait_for_tick"></a>**<font color="#7fb800">wait_for_tick</font>**(<font color="#00a6ed">**self**</font>, <font color="#00a6ed">**seconds**=10.0</font>)  
This method is used in [__asynchronous__ mode](https://[carla.readthedocs.io](#carla.readthedocs.io)/en/latest/adv_synchrony_timestep/). It makes the client wait for a server tick. When the next frame is computed, the server will tick and return a snapshot describing the new state of the world.  
    - **Parameters:**
        - `seconds` (_float_) – Maximum time in seconds the server should wait for a tick. It is set to `10.0` by default.  
    - **Return:** _[carla.WorldSnapshot](#carla.WorldSnapshot)_

huan0808, 2020-09-24:
I think it's good now,thank you :)

closed 2020-09-24 by huan0808

yshen47, 2020-11-07:
I also find it's misleading. could you update the documentation in the latest version? it is still like before

Lamfurst, 2023-06-25:
Hi everyone,

I have a question regarding the usage of the wait_for_tick() function in synchronous mode. After going through the discussion here, I understand that it's generally advised not to use wait_for_tick() in synchronous mode. However, I came across a video [here](https://www.youtube.com/watch?v=nyyTLmphqY4), between 0:48 - 2:22, where wait_for_tick() is used in synchronous mode. I'm a bit confused about this discrepancy, and I would appreciate some clarification.

From my understanding, wait_for_tick() should typically be used in asynchronous mode to obtain a snapshot that can be later used in a callback function registered with on_tick(). This allows for handling and processing of the simulation data asynchronously. Could someone please correct me if I'm mistaken?

I'm curious to know if there are specific scenarios where using wait_for_tick() in synchronous mode is acceptable or if there might be any drawbacks or implications of doing so. Any insights or clarification would be greatly appreciated.