carla-simulator/carla #3284: Suggestion to add "sky-label" to default Carla Labels

dnns92, 2020-09-11:
Dear Contributors,

as suggested here: https://github.com/carla-simulator/carla/issues/2628 and also here https://github.com/carla-simulator/carla/issues/1617, one can add the sky label by adding the respective string comparison to a couple of header/cpp files and compile carla from scratch. However, this becomes very tidious as I have to do this every time a new carla version gets released. Additionally, building takes a lot of time itself. Therefore, right now, I'm using a seperate depth camera in the same place as the segmentation camera and label the sky in the segmentation image by the distance of the depth camera. 

I think it's reasonable to add a sky-label permenantly to carla. Having the sky as an extra labels adds compatibility to the available research segmentation datasets such as cityscapes. Futher, it makes segmented images way easier to understand without looking at the corresponding RGB-image. 
I think it's a reasonable improvement with litte backlashes for users to add the sky label permanantly. Users that dont want it can still set the Sky to "unknown" in the postprocessing easily.

Cheers, Dennis

germanros1987, 2020-09-24:
Noted.

XGodina, 2020-09-29:
Hi @dnns92 ,

We've implemented this tag in this latest release just as we have improved our semantic segmentation making it more detailed. I encourage you to try this release 9.10 to see what you think.

closed 2020-12-25 by stale[bot]