carla-simulator/carla #3285: Building CARLA 0.9.9.4 from source always fails

ndeshp2s, 2020-09-11:
Hello, I am trying to build carla 0.9.9.4 from source since a few days now, however, it always gives an error after running "make package".                                                                       

The error is as follows:                                                                                                                                                                                                                                                                                    ERROR: Cook failed.
 (see /home/niranjan/Library/Logs/Unreal Engine/LocalBuildLogs/Log.txt for full exception trace)
AutomationTool exiting with ExitCode=25 (Error_UnknownCookFailure)
RunUAT ERROR: AutomationTool was unable to run successfully.
Util/BuildTools/Linux.mk:16: recipe for target 'package' failed
make: *** [package] Error 25
                                                                                                                                                                                                                                                                                                                                  I tested the unreal engine installation and it works fine. 
But even with "make launch" command I cannot run the server of carla. As soon as I press the play button in carlaUE it freezes for a while and then crashes. 

I tried this installation on 2 different systems with fresh ubuntu installation. But always getting same error.                                                                 Does anybody have any idea/solution for this please?

germanros1987, 2020-09-24:
Which version of UE4 are you using?
@bernatx any feedback?

bernatx, 2020-09-24:
Checking the file  /home/niranjan/Library/Logs/Unreal Engine/LocalBuildLogs/Log.txt could help in getting clues.
It seems the process of cooking the package fails. Do you have any custom asset or map inside?

closed 2020-12-25 by stale[bot]

xuisshoe, 2021-03-18:
Why is this closed.....