carla-simulator/carla #3286: Can I add a new segmentation tag in windows version? How can I do that?

pangxiao2013, 2020-09-11:


germanros1987, 2020-09-24:
@sergi-e could you please point to the documentation page that explains this?

sergi-e, 2020-10-01:
Hi, @pangxiao2013 !

Here are some instructions on how to create a semantic tag. First of all, you have to work with a CARLA build, and be updated with the latest changes in `dev`. 

### 1. Create the tag ID

__Open `ObjectLabel.h`__ in `LibCarla/source/carla/rpc`. Add your new tag by the end of the enum using the same formatting as the rest.  

    Tags do not have to appear in order. However, it is good practice to list them in order. 

### 2. Create the UE folder for assets

__Open the Unreal Engine Editor__ and go to `Carla/Static`. Create a new folder named as your tag. 

    The UE folder and the tag do not necessarily have to be named the same. However, it is good practice to do so.  

### 3. Create two-way correspondence between UE and the code tag

__3.1. Open `Tagger.cpp`__ in `Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game`. Go to __`GetLabelByFolderName`__ Add the your tag by the end of the list.  

__3.2. Go to `GetTagAsString`__ in the same `Tagger.cpp`. Add the new tag by the end of the switch.  

### 4. Define a color code

__Open `CityScapesPalette.h`__ in `LibCarla/source/carla/image`. Add the color code of your new tag by the end of the array. 

[image]

    The position in the array must correspond with the tag ID, in this case, `23u`. 

### 5. Add the tagged elements

The new semantic tag is ready to be used. Only the elements stored inside the UE folder of a tag are tagged as such. Move or import the corresponding elements to the new folder, in order for the to be tagged properly. 

---

### 6. Add a tag to [carla.CityObjectLabel](python_api.md#carla.CityObjectLabel)

This step is not directly related with semantic segmentation. However, these tags can be used to filter the bounding box query in [carla.World](python_api.md#carla.World). In order to do this, the tag must be added to the [carla.CityObjectLabel](python_api.md#carla.CityObjectLabel) enum in the PythonAPI.  

__Open `World.cpp`__ in `carla/PythonAPI/carla/sourc/libcarla` and add the new tag by the end of the enum.

pangxiao2013, 2020-10-08:
> Hi, @pangxiao2013 !
>
> Here are some instructions on how to create a semantic tag. First of all, you have to work with a CARLA build, and be updated with the latest changes in `dev`.
>
> ### 1. Create [...]

thanks a lot. But it is still difficiult for me as a new carla user. I hope for segmentation label in carla platform, developers can make standard rich labels like cityscape, more specificly, vechiles can be divided into car, trunk, bicycle and etc. Pedestrian can be devided into riders and etc. BTY thanks a lot.

closed 2020-10-15 by sergi-e