carla-simulator/carla #3291: [Co-simulation error]'Connection' object has no attribute '_socket'

zwbai, 2020-09-14:
Hi,

I want to run the co-simulation with Sumo and I followed the tutorial 'Spawn NPCs controlled by SUMO'. I started the carla town 01 and then run the following code:
`python3 spawn_npc_sumo.py  --tls-manager sumo --sumo-gui --step-length 0.1 -n 50`

The problem is: the co-simulation is all the fine at the beginning. However, after a few time (10-30 seconds), the sumo will stop and show the error massage which is shown as bellow. 

[image]

It is noticed that when the error occurs, the carla server and manual_control.py are still running. So I want to know what's the problem and how to solve it.

Thanks a lot!

joel-mb, 2020-09-21:
Hi @zwbai 
Are you able to reproduce the error always? Is it happening with all the towns? I would say this may be related to a bug inside SUMO.

germanros1987, 2020-09-24:
@Axel1092 in case additional support is needed.

zwbai, 2020-09-24:
Hi @joel-mb @germanros1987  ,

Thank you for your attention. I have tested in the several towns. The problem happened in all of them. 
One thing that may be valuable for solving this problem is that: I found that when I increased the step length (e.g., --step-length 0.15) the SUMO Co-simulation will successfully runing with longer time before it shut down (it will still shut down all the time and show the same error massages) than the default step length (0.05).
Hope that the information is valuable.

Thank you again!

joel-mb, 2020-09-25:
Hi @zwbai !
Thank you for the information! I'm looking into it.
Is this only happening with `spawn_npc_sumo`? Have you reproduced it using the `run_syncrhonization` script?

closed 2020-12-25 by stale[bot]