carla-simulator/carla #3294: Carla - Window's installation fails at "make launch"

chandeep11, 2020-09-14:
I'm trying to install Carla 0.9.8 (due to project dependency ) in Window's 10. I've installed VS2017 community edition and UE 4.22.3 and all other dependencies mentioned in the Build on Windows doc page. 

When I try to run **make launch**, it fails with the following error.

```
D:\carla>make launch

-[Setup]: [Batch params]: --boost-toolset msvc-14.1
-[Setup]: Asynchronous jobs:  16
-[Setup]: Boost toolset:      msvc-14.1
-[Setup]: Install directory:  "D:\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "D:\carla\Build\"
    -[install_zlib]: Not downloading zlib because already exists the folder "D:\carla\Build\zlib-source".
-- The C compiler identification is MSVC 19.16.27043.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx64/x64/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx64/x64/cl.exe - broken
CMake Error at C:/Program Files/CMake/share/cmake-3.18/Modules/CMakeTestCCompiler.cmake:66 (message):
  The C compiler

 "C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx64/x64/cl.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: D:/carla/Build/zlib-source/build/CMakeFiles/CMakeTmp

    Run Build Command(s):nmake /nologo cmTC_cae55\fast &&       "C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX64\x64\nmake.exe"  -f CMakeFiles\cmTC_cae55.dir\build.make /nologo -L                  CMakeFiles\cmTC_cae55.dir\build
    Building C object CMakeFiles/cmTC_cae55.dir/testCCompiler.c.obj
        "C:\PROGRA~2\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\Hostx64\x64\cl.exe" @C:\Users\Demo-2\AppData\Local\Temp\nm35C3.tmp
    testCCompiler.c
    Linking C executable cmTC_cae55.exe
        "C:\Program Files\CMake\bin\cmake.exe" -E vs_link_exe --intdir=CMakeFiles\cmTC_cae55.dir --rc="C:\PROGRA~2\Windows Kits\8.1\bin\x64\rc.exe" --mt="C:\PROGRA~2\Windows Kits\8.1\bin\x64\mt.exe" --manifests -- "C:\PROGRA~2\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\Hostx64\x64\link.exe" /nologo @CMakeFiles\cmTC_cae55.dir\objects1.rsp @C:\Users\Demo-2\AppData\Local\Temp\nm3603.tmp
    LINK Pass 1: command "C:\PROGRA~2\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\Hostx64\x64\link.exe /nologo @CMakeFiles\cmTC_cae55.dir\objects1.rsp /out:cmTC_cae55.exe /implib:cmTC_cae55.lib /pdb:D:\carla\Build\zlib-source\build\CMakeFiles\CMakeTmp\cmTC_cae55.pdb /version:0.0 /machine:x64 /debug /INCREMENTAL /subsystem:console kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib /MANIFEST /MANIFESTFILE:CMakeFiles\cmTC_cae55.dir/intermediate.manifest CMakeFiles\cmTC_cae55.dir/manifest.res" failed (exit code 1104) with the following output:
    LINK : fatal error LNK1104: cannot open file 'ucrtd.lib'
    NMAKE : fatal error U1077: '"C:\Program Files\CMake\bin\cmake.exe"' : return code '0xffffffff'
    Stop.
    NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX64\x64\nmake.exe"' : return code '0x2'
    Stop.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:4 (project)

-- Configuring incomplete, errors occurred!
See also "D:/carla/Build/zlib-source/build/CMakeFiles/CMakeOutput.log".
See also "D:/carla/Build/zlib-source/build/CMakeFiles/CMakeError.log".

    -[install_zlib]: [CMAKE ERROR] An error ocurred while executing cmake command.
    -[install_zlib]: [CMAKE ERROR] Possible causes:
    -[install_zlib]:                - Make sure "CMake" is installed.
    -[install_zlib]:                - Make sure it is available on your Windows "path".
    -[install_zlib]:                - Make sure you have cmake 3.12.4 or higher installed.
    -[install_zlib]: Exiting with error...

-[Setup]:
   Ok, and error ocurred, don't panic!
   We have different platforms where you can find some help :)

   - Make sure you have read the documentation:
       http://carla.readthedocs.io/en/latest/how_to_build_on_windows/

   - If the problem persists, you can ask on our Github's "Building on Windows" issue:
       https://github.com/carla-simulator/carla/issues/21

   - Or just use our Discord channel!
       We'll be glad to help you there :)
       https://discord.gg/42KJdRj
make: *** [setup] Error 1

D:\carla> 
```

I've also tried updating the **CMakeLists.txt** file to use the installed CMake, which also reports the same error.

```
cmake_minimum_required(VERSION 3.5.1)

set(CMAKE_C_COMPILER "C:/Program Files/CMake/bin/cmake.exe")
set(CMAKE_CXX_COMPILER "C:/Program Files/CMake/bin/cmake.exe")

project(CARLA)
include("Build/CMakeLists.txt.in")
add_subdirectory("LibCarla/cmake")
```

CMake is installed and is added to the PATH variable.

```
D:\carla>where cmake
C:\Program Files\CMake\bin\cmake.exe
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe
```

Tried the solutions provided in FAQ which also seems to be not working. Kindly provide any solution to procced further.

seuwcs, 2020-09-16:
I met the same problems and can't find a solution.

germanros1987, 2020-09-24:
@sergi-e could you provide some help and perhaps update the documentation?

YDSh510, 2020-09-25:
Hi Carla Team,

Any suggestion or fix on the above issue (@chandeep11) , as I am also struck with it.

chandeep11, 2020-09-28:
@sergi-e Could you please update on this as I've stuck with this for more than 2 weeks.! I can't proceed further due to this issue.

nagasanjay, 2020-10-10:
Please make sure you are running `make launch` from the `x64 native tools command prompt for vs 2017` and not from regular `cmd`. 
Also ensure you've installed `c++ and .NET workloads` from the vivsual studio installer.

EchoYJJ, 2020-12-20:
Hi, @chandeep11  I also meet the same issue, have you found a solution or an alternative method? Thanks a lot!

closed 2021-07-21 by stale[bot]