carla-simulator/carla #3298: Carla client crash with ray

dianchen96, 2020-09-16:
Hello!

I have been trying to collect data using the ray library, where I have several parallel actors each having a client connecting to each server (e.g 3 servers and 3 ray actors with world ports 2000,4000,6000, and traffic ports 1999,2999, and 3999).

However it seems to arbitrarily crash with the following error trace

```
(pid=26449) *** Aborted at 1600223752 (unix time) try "date -d @1600223752" if you are using GNU date ***
(pid=26449) PC: @                0x0 (unknown)
(pid=26449) *** SIGSEGV (@0x7f1e2746ae30) received by PID 26449 (TID 0x7f0f5d29e700) from PID 658943536; stack trace: ***
(pid=26449)     @     0x7f1e37d86f20 (unknown)
(pid=26449)     @     0x55646d4d9347 pymalloc_alloc.isra.1
(pid=26449)     @     0x55646d57b4fc _PyObject_Malloc
(pid=26449)     @     0x55646d595bb8 PyUnicode_New
(pid=26449)     @     0x55646d5cf35f PyUnicode_FromWideChar
(pid=26449)     @     0x55646d5dfecf unicode_decode_locale
(pid=26449)     @     0x55646d5e0515 PyErr_SetFromErrnoWithFilenameObjects
(pid=26449)     @     0x7f1e35f15a05 sock_call_ex.cold.33
(pid=26449)     @     0x7f1e35f1ba10 sock_recv_into
(pid=26449)     @     0x55646d640420 _PyEval_EvalFrameDefault
(pid=26449)     @     0x55646d58fdb2 _PyEval_EvalCodeWithName
(pid=26449)     @     0x55646d5afb63 _PyFunction_FastCallKeywords
(pid=26449)     @     0x55646d639ae5 _PyEval_EvalFrameDefault
(pid=26449)     @     0x55646d58fdb2 _PyEval_EvalCodeWithName
(pid=26449)     @     0x55646d5afb63 _PyFunction_FastCallKeywords
(pid=26449)     @     0x55646d5f6335 call_function
(pid=26449)     @     0x55646d63aaee _PyEval_EvalFrameDefault
(pid=26449)     @     0x55646d5af767 _PyFunction_FastCallKeywords
(pid=26449)     @     0x55646d5f6335 call_function
(pid=26449)     @     0x55646d639d01 _PyEval_EvalFrameDefault
(pid=26449)     @     0x55646d58fdb2 _PyEval_EvalCodeWithName
(pid=26449)     @     0x55646d5afb63 _PyFunction_FastCallKeywords
(pid=26449)     @     0x55646d5f6335 call_function
(pid=26449)     @     0x55646d63aaee _PyEval_EvalFrameDefault
(pid=26449)     @     0x55646d58fdb2 _PyEval_EvalCodeWithName
(pid=26449)     @     0x55646d5afb63 _PyFunction_FastCallKeywords
(pid=26449)     @     0x55646d5f6335 call_function
(pid=26449)     @     0x55646d63aaee _PyEval_EvalFrameDefault
(pid=26449)     @     0x55646d58fdb2 _PyEval_EvalCodeWithName
(pid=26449)     @     0x55646d5afb01 _PyFunction_FastCallKeywords
(pid=26449)     @     0x55646d5f6335 call_function
(pid=26449)     @     0x55646d639d01 _PyEval_EvalFrameDefault
```

I am wondering if anyone has seen this before, or could share some lights on why this happens? Thanks a lot!

dianchen96, 2020-09-16:
Closing the issue, it seems to be caused by me not freeing the streaming port of previous runs. My bad!

dianchen96, 2020-09-24:
Reopening this as

bernatx, 2020-10-30:
Hi,
Do you have any small script that can reproduce the error? 
From the stack is hard to say what can be the problem. 
It seems a buffer allocation problem where the socket is trying to save the received data.

closed 2021-04-27 by stale[bot]