carla-simulator/carla #3299: Make launch Error while compiling Carla

Malathi15, 2020-09-16:
Hi,

My System Specifications:
OS : Ubuntu 18.04 (64 bit)
Memory : 15.7 GiB
Processor : AMD® Ryzen 7 3800x 8-core processor × 16 
Graphics : GeForce GTX 1050 Ti/PCIe/SSE2

I am trying to build Carla Simulator followed the instruction given in this link
 https://carla.readthedocs.io/en/latest/build_linux/
I have downloaded the Carla from this link https://github.com/carla-simulator/carla
I have followed all the steps mentioned in the instruction.

While running `sudo make launch` I got an following error
```
Setup.sh: llvm-8.0 already installed.
Setup.sh: boost-1.72.0-c8 already installed.
Setup.sh: rpclib-v2.2.1_c3-c8 already installed.
Setup.sh: gtest-1.8.1-c8 already installed.
Setup.sh: recast-cdce4e-c8 already installed.
Setup.sh: Libpng already installed.
Setup.sh: CARLA version 0.9.9.4-322-g5bdd65ca.
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
BuildLibCarla.sh: Building LibCarla "Server.Release" configuration.
ninja: no work to do.
[0/1] Install the project...
-- Install configuration: "Server"
BuildLibCarla.sh: Success!
BuildCarlaUE4.sh: ERROR: UE4_ROOT is not defined, or points to a non-existant directory, please set this environment variable.
Util/BuildTools/Linux.mk:7: recipe for target 'launch' failed
make: *** [launch] Error 2
```
Please check the screenshot
[image]
My CMake version is 3.15.2
I have already build UnrealEngine_4.24 successfuly. I have also check UE4_ROOT using this following command
`~/carla$ printenv | grep UE4_ROOT`
got this output
`UE4_ROOT=/home/ioz/UnrealEngine_4.24`

I have add
`UE4_ROOT=/home/ioz/UnrealEngine_4.24`
in `Util/BuildTools/BuildCarlaUE4.sh `
It is not solve the problem.
Rebuild the UnrealEngine also not solve this problem

Please help me to solve this

Thanks in Advance
Malathi K

germanros1987, 2020-09-24:
@bernatx could you please help with the build?

bernatx, 2020-09-24:
We use the **make** commands without sudo, but I think it will not do any difference.
Can you show how is the structure of the UnrealEngine_4.24? Is it inside another folder?
It seems to me that the environment variable UE4_ROOT is correct.

qhaas, 2020-10-02:
> We use the **make** commands without sudo, but I think it will not do any difference.

[Epic says not to use root/sudo with UnrealEngine](https://answers.unrealengine.com/questions/913405/build-from-source-refusing-root-privileges.html?sort=oldest)
`do not build it as root. Parts of Unreal engine need to run during the build and they will not run with superuser privileges.`

closed 2020-12-25 by stale[bot]