carla-simulator/carla #3303: Testing PR #3301, Retrieve BBs of all the elements of the level

doterop, 2020-09-16:
This issue is intended to track #3301 .

Need to test:

- All the geometries have a BB.
- The BBs are not duplicated for the same geometry.
- All the BB are in the right position, with the proper rotation and the size needed to fit the geometry.
- The object type filter doesn't return objects of another type.
- The object type filter doesn't return all the objects of the level for that type.

Example:

```
bbs = world.get_level_bbs(carla.CityObjectLabel.TrafficLight)
debug = world.debug

print("Num BBs = " + str(len(level_bbs)))

for bb in level_bbs :
    debug.draw_box(bb, bb.rotation, 0.05, carla.Color(255,0,0,0))
```

marcgpuig, 2020-09-22:
- Looks like the spline based meshes are wrongly transformed
- This code is duplicated:
https://github.com/carla-simulator/carla/blob/f1dd2d77fa908af76215ba7a640fa5543b42b64d/LibCarla/source/carla/geom/Rotation.h#L90-L116
- This should be extended with the new rotation:
https://github.com/carla-simulator/carla/blob/f1dd2d77fa908af76215ba7a640fa5543b42b64d/PythonAPI/carla/source/libcarla/Geom.cpp#L65-L70
- Also the function should send the semantic tag id of the contained object, but this can be for the next iteration.

doterop, 2020-09-24:
Thanks for point that out @marcgpuig .
I already fixed the spline issue, the duplication on the Rotation.h and output of the BB.

doterop, 2020-09-24:
Updated this issue with the new PR #3360

tinmodeHuang, 2020-10-14:
> This issue is intended to track ##3360 (previous PR: ~#3301~).
> 
> Need to test:
> 
> * All the geometries have a BB.
> * The BBs are not duplicated for the same geometry.
> * All the BB are in the right position, with the proper rotation and the size needed to fit the geometry.
> * The object type filter doesn't return objects of another type.
> * The object type filter doesn't return all the objects of the level for that type.
> 
> Example:
> 
> ```python
> bbs = world.get_level_bbs(carla.CityObjectLabel.TrafficLight)
> debug = world.debug
> 
> print("Num BBs = " + str(len(level_bbs)))
> 
> for bb in level_bbs :
>     debug.draw_box(bb, bb.rotation, 0.05, carla.Color(255,0,0,0))
> ```
hi ! @doterop. do you mind if I ask a question, I wonder if the index of the same vehicle remains the same in the return got after calling world.get_level_bbs(carla.CityObjectLabel.TrafficLight) every time in the same simulation?

doterop, 2020-10-19:
Hi @tinmodeHuang ,

Sorry but I don't really understand the question because I don't see the relation between vehicles and traffic lights in the BB call.

If you are concern about the order of the retrieved BBs, meanwhile there is no new spawned or destroyed actors it should be the same.

tinmodeHuang, 2020-10-19:
> Hi @tinmodeHuang ,
> 
> Sorry but I don't really understand the question because I don't see the relation between vehicles and traffic lights in the BB call.
> 
> If you are concern about the order of the retrieved BBs, meanwhile there is no new spawned or destroyed actors it should be the same.

thanks, that should be `carla.CityObjectLabel.Vehicle`

doterop, 2020-10-19:
Nice

tinmodeHuang, 2020-10-19:
> Nice

have you the answer to that question?

doterop, 2020-10-19:
Yes but in the other issue

closed 2021-01-26 by germanros1987