carla-simulator/carla #3308: Bind Carla server to localhost (127.0.0.1)

aleallievi, 2020-09-16:
We are using CARLA on a workstation within a university network. Because of our set-up, we were notified that the CARLA server listening on port 0.0.0.0 is a security vulnerability. How can we bind the server to port 127.0.0.1 instead, so as to ensure all client server communication occurs locally via the loopback ports? Your support is appreciated.

germanros1987, 2020-09-24:
@bernatx could you comment on this?

germanros1987, 2020-09-24:
One suggestion would be to use the CARLA docker. That will protect from this issue.

aleallievi, 2020-09-24:
@germanros1987 thanks for the feedback - yes we are getting around the issue via Docker. Feel free to close the issue, though a handy way of ensuring the server is communicating via loopback ports would be a nice security feature.

closed 2020-12-25 by stale[bot]