carla-simulator/carla #3310: Compile LibCarla to work with RSS Error

liuyandong1988, 2020-09-17:
I want to use the rss sensor in Carla. When i build the RSS  (make LibCarla.client.rss), some errors happened. I have installed the dependencies based on the carla simulator docs.

Have installed:
Ubuntu 16.04
python 3.7
Unreal Engine 4.24
carla 0.9.9

CMake Error at LibCarla/cmake/client/CMakeLists.txt:11 (find_package):
  Could not find a package configuration file provided by "spdlog" with any
  of the following names:

    spdlogConfig.cmake
    spdlog-config.cmake

  Add the installation prefix of "spdlog" to CMAKE_PREFIX_PATH or set
  "spdlog_DIR" to a directory containing one of the above files.  If "spdlog"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!

germanros1987, 2020-09-24:
@fpasch could you please take a look at this issue?

closed 2020-12-25 by stale[bot]

NikhilKalghatgi, 2021-05-18:
Hi, I am facing this issue too, please help.