carla-simulator/carla #3317: Using manual control using ros and not ros bridge

poornimajd, 2020-09-18:
Thanks for the great work!
My aim is to manually control the car without using keyboard keys or pygame.My problem is how do I give a ros command so that the actor(car) takes the value and moves accordingly.I am trying not to use ros bridge because I am a beginner in ros and I was getting a lot of errors while trying the same.Is there a way to publish a value of say throttle using ros(vanilla) and using carla to subscribe to it and move the car accordingly.
Basically an alternative to 
https://github.com/carla-simulator/carla/blob/dev/PythonAPI/examples/manual_control.py#L419
instead of pygame it will subscribe to ros.
Thank you

germanros1987, 2020-09-24:
@joel-mb should be able to help with this question.

joel-mb, 2020-09-25:
Hi @poornimajd!
I strongly recommend to you using the `ros-bridge`. There you have an alternative to the manual control script that is using only ros (https://github.com/carla-simulator/ros-bridge/blob/master/carla_manual_control/src/carla_manual_control/carla_manual_control.py).

However, If you don't want to use this node, you can also publish the vehicle command message in the required topic (`/carla/ego_vehicle/vehicle_control_cmd`). Here you have an example using `rostopic pub`

1. Start the `ros-bridge` node and the `ego-vehicle` node:
```
roslaunch carla_ros_bridge carla_ros_bridge.launch
roslaunch carla_ego_vehicle carla_example_ego_vehicle.launch
```
2. Publish the vehicle command: 
```
rostopic pub -r 1 /carla/ego_vehicle/vehicle_control_cmd carla_msgs/Carla
  EgoVehicleControl "header:
    seq: 0
    stamp: {secs: 0, nsecs: 0}
    frame_id: ''
  throttle: 1.0
  steer: 0.0
  brake: 0.0
  hand_brake: false
  reverse: false
  gear: 0
  manual_gear_shift: false"
```
Lastly, if you don't want to use the `ros-bridge` at all, you should create a node containing a carla client. The node should be subscribed to the vehicle command topic and apply the control to the vehicle using the carla PythonAPI (https://carla.readthedocs.io/en/latest/python_api/)

I hope this helps!

closed 2020-09-30 by joel-mb