carla-simulator/carla #3318: How to change the PID controller of Traffic Manager?

dhiegomaga, 2020-09-18:
I've noticed that the cars in autopilot move in a very robotic and unnatural manner, and I'd like to make them more human like. I know you can set some values, such as % of speed limit to drive, etc. 

However, is there a way to change the PID constant values for a given vehicle actor? For instance, they always seem to "brake" with the same intensity, even though I changed the "distance_to_leading_vehicle" to the minimum, they still brake smoothly, etc, and it would be interesting to have them brake very suddenly or very slowly, etc. Also, it would be interesting to be able to tweak the amount of steering when making a turn, so that some vehicles would turn more abruptly than others. 

Am I correct to assume these configurations could be changed by simply changing the PID values? And how could it be changed?

jackbart94, 2020-09-21:
Hi,

The values for the PID constants are in ``TrafficManager.cpp``, however you could also be interested in the huge variety of constants in the file ``Constants.h`` that manage for example at what distance vehicles start braking, what is the maximum they brake at, etc.

dhiegomaga, 2020-09-23:
> Hi,
> 
> The values for the PID constants are in `TrafficManager.cpp`, however you could also be interested in the huge variety of constants in the file `Constants.h` that manage for example at what distance vehicles start braking, what is the maximum they brake at, etc.

Where exactly in the [TrafficManager.cpp](https://github.com/carla-simulator/carla/blob/dev/LibCarla/source/carla/trafficmanager/TrafficManager.cpp)? 

I looked into [Constants.h](https://github.com/carla-simulator/carla/blob/dev/LibCarla/source/carla/trafficmanager/Constants.h) and found the following: 

```
 namespace PID {
static const float MAX_THROTTLE = 0.7f;
static const float MAX_BRAKE = 1.0f;
static const float VELOCITY_INTEGRAL_MAX = 5.0f;
static const float VELOCITY_INTEGRAL_MIN = -5.0f;
static const float DT = 0.05f;
static const float INV_DT = 1.0f / DT;
}
```

But these constants don't seem to control the brake _intensity, acceleration, etc_, but rather the max values, right?

closed 2020-12-13 by stale[bot]