carla-simulator/carla #3320: Make PythonAPI fatal error C1083: Cannot open include file 'OSM2ODR.h' No such file or directory:

jaimindp, 2020-09-18:
Hi, I'm building on Windows and make launch runs fine. When I run make PythonAPI, I receive this error log. it finishes with:

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "C:\Users\User\Documents\carla\PythonAPI\carla\dist"!

Yet there is no directory /dist created. Has anyone else come across this issue?

Here is the full error log:

-[Setup]: [Batch params]: --boost-toolset msvc-14.1
-[Setup]: Asynchronous jobs:  8
-[Setup]: Boost toolset:      msvc-14.1
-[Setup]: Install directory:  "C:\Users\User\Documents\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "C:\Users\User\Documents\carla\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "C:\Users\User\Documents\carla\Build\zlib-install\" if you want to force a rebuild.
    -[install_zlib]: Exiting...
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "C:\Users\User\Documents\carla\Build\" --zlib-install-dir "C:\Users\User\Documents\carla\Build\zlib-install\"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "C:\Users\User\Documents\carla\Build\libpng-1.2.37-install\" if you want to force a rebuild.
    -[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "C:\Users\User\Documents\carla\Build\"
    -[install_rpclib]: A rpclib installation already exists.
    -[install_rpclib]: Delete "C:\Users\User\Documents\carla\Build\rpclib-install\" if you want to force a rebuild.
    -[install_rpclib]: Exiting...
-[Setup]: Installing Google Test...
    -[install_gtest]: [Batch params]: --build-dir "C:\Users\User\Documents\carla\Build\"
    -[install_gtest]: A Google Test installation already exists.
    -[install_gtest]: Delete "C:\Users\User\Documents\carla\Build\gtest-install\" if you want to force a rebuild.
    -[install_gtest]: Exiting...
-[Setup]: Installing "Recast & Detour"...
    -[install_recast]: [Batch params]: --build-dir "C:\Users\User\Documents\carla\Build\"
    -[install_recast]: A "Recast & Detour" installation already exists.
    -[install_recast]: Delete "C:\Users\User\Documents\carla\Build\recast-cdce4e-install\" if you want to force a rebuild.
    -[install_recast]: Exiting...
-[Setup]: Installing Boost...
    -[install_boost]: [Batch params]: --build-dir "C:\Users\User\Documents\carla\Build\" --toolset msvc-14.1 --version 1.72.0 -j 8
    -[install_boost]: A Boost installation already exists.
    -[install_boost]: Delete "C:\Users\User\Documents\carla\Build\boost-1.72.0-install\" if you want to force a rebuild.
    -[install_boost]: Exiting...
        1 file(s) copied.
        1 file(s) copied.
-[Setup]: Installing Xercesc...
    -[install_xercesc]: [Batch params]: --build-dir "C:\Users\User\Documents\carla\Build\"
    -[install_xercesc]: A xerces installation already exists.
    -[install_xercesc]: Delete "C:\Users\User\Documents\carla\Build\xerces-c-3.2.3-install\" if you want to force a rebuild.
    -[install_xercesc]: Exiting...
-[Setup]: Creating "CMakeLists.txt.in"...
-[Setup]:
   ###########
   # SUCCESS #
   ###########

   IMPORTANT!

   All the CARLA library dependences should be installed now.
   (You can remove all "*-src" folders in C:\Users\User\Documents\carla\Build\ directory)

   You only need the ASSET PACK with all the meshes and textures.

   This script provides the assets for CARLA 0.9.10:
   You can download the assets from here:

       http://carla-assets.s3.amazonaws.com/20200731_3e91044.tar.gz

   Unzip it in the "C:\Users\User\Documents\carla\Unreal\CarlaUE4\Content\Carla\" folder.
   If you want another version, search it in C:\Users\User\Documents\carla\Util\ContentVersions.txt.

[... 149 lines cut ...]
running bdist_egg
running egg_info
writing source\carla.egg-info\PKG-INFO
writing dependency_links to source\carla.egg-info\dependency_links.txt
writing top-level names to source\carla.egg-info\top_level.txt
reading manifest file 'source\carla.egg-info\SOURCES.txt'
writing manifest file 'source\carla.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
running install_lib
running build_py
running build_ext
building 'carla.libcarla' extension
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Idependencies/include -IC:\Users\User\AppData\Local\Programs\Python\Python37\include -IC:\Users\User\AppData\Local\Programs\Python\Python37\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\cppwinrt" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\cppwinrt" /EHsc /Tpsource/libcarla/libcarla.cpp /Fobuild\temp.win-amd64-3.7\Release\source/libcarla/libcarla.obj /experimental:external /external:I dependencies/include/system /DBOOST_ALL_NO_LIB /DBOOST_PYTHON_STATIC_LIB /DBOOST_ERROR_CODE_HEADER_ONLY /D_WIN32_WINNT=0x0600 /DHAVE_SNPRINTF /DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true
libcarla.cpp
dependencies/include\png.h(522): warning C4005: 'int_p_NULL': macro redefinition
dependencies/include\carla/image/ImageIOConfig.h(48): note: see previous definition of 'int_p_NULL'
dependencies/include/system\boost/gil/io/path_spec.hpp(75): warning C4996: 'wcstombs': This function or variable may be unsafe. Consider using wcstombs_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.
C:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt\stdlib.h(1015): note: see declaration of 'wcstombs'
dependencies/include/system\boost/gil/io/path_spec.hpp(120): warning C4996: 'wcstombs': This function or variable may be unsafe. Consider using wcstombs_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.
C:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt\stdlib.h(1015): note: see declaration of 'wcstombs'
dependencies/include/system\boost/gil/io/path_spec.hpp(129): warning C4996: 'wcstombs': This function or variable may be unsafe. Consider using wcstombs_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.
C:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt\stdlib.h(1015): note: see declaration of 'wcstombs'
c:\users\user\documents\carla\pythonapi\carla\source\libcarla\OSM2ODR.cpp(7): fatal error C1083: Cannot open include file: 'OSM2ODR.h': No such file or directory

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "C:\Users\User\Documents\carla\PythonAPI\carla\dist"!

H-Whisky, 2020-09-22:
I also meet the issues when I update the code to master. I can run "make launch" and run a simulator by the Unreal Edior. But when I run "make PythonAPI" to get the corresponding *.egg file, I meet the following error:
```
d:\workspace\carla\pythonapi\carla\source\libcarla\OSM2ODR.cpp(7): fatal error C1083: Cannot open include file: “OSM2ODR.h”: No such file or directory
error: command 'D:\\software\\VisualStudio2017\\VC\\Tools\\MSVC\\14.16.27023\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "D:\workspace\carla\PythonAPI\carla\dist"!
```
I see the error notification, so I go to the carla/build, and I see that the **xerces-c-3.2.3-install** folder is null, and the **xerces-c-3.2.3-source** has been download, I dont't know how to make the **xerces-c-3.2.3-install** really install. Need help too. Thank you!

germanros1987, 2020-09-24:
@Axel1092 could you please provide some help with this issue?

Axel1092, 2020-09-24:
HI @H-Whisky, try removing all `xerces*` files and folders from the carla/build folder. Then run `make PythonAPI` again. If you still get an error please share again with us the full console log of this run.

VKG5, 2020-09-25:
> HI @H-Whisky, try removing all `xerces*` files and folders from the carla/build folder. Then run `make PythonAPI` again. If you still get an error please share again with us the full console log of this run.

Hey, I was getting the same error so I deleted the 'xerces*' folders and filers in the 'carla/build' folder. Then ran the command but I'm still getting the same error.

## Error

-- Configuring incomplete, errors occurred!
See also "F:/Programs/Project/Carla/Build/osm2odr-visualstudio/CMakeFiles/CMakeOutput.log".
Microsoft (R) Build Engine version 15.9.21+g9802d43bc3 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

MSBUILD : error MSB1009: Project file does not exist.
Switch: install.vcxproj
-[BuildOSM2ODR]: OSM2ODR has been successfully installed in "F:\Programs\Project\Carla\PythonAPI\carla\dependencies\"
-[BuildPythonAPI]: [Batch params]: --py3
Building Python API for Python 3.
compiling:
  - source/libcarla/libcarla.cpp
  - dependencies/include/carla\Buffer.cpp
  - dependencies/include/carla\Exception.cpp
  - dependencies/include/carla\FileSystem.cpp
  - dependencies/include/carla\StringUtil.cpp
  - dependencies/include/carla\client\Actor.cpp
  - dependencies/include/carla\client\ActorAttribute.cpp
  - dependencies/include/carla\client\ActorBlueprint.cpp
  - dependencies/include/carla\client\ActorList.cpp
  - dependencies/include/carla\client\BlueprintLibrary.cpp
  - dependencies/include/carla\client\DebugHelper.cpp
  - dependencies/include/carla\client\Junction.cpp
  - dependencies/include/carla\client\LaneInvasionSensor.cpp
  - dependencies/include/carla\client\Light.cpp
  - dependencies/include/carla\client\LightManager.cpp
  - dependencies/include/carla\client\Map.cpp
  - dependencies/include/carla\client\ServerSideSensor.cpp
  - dependencies/include/carla\client\TimeoutException.cpp
  - dependencies/include/carla\client\TrafficLight.cpp
  - dependencies/include/carla\client\TrafficSign.cpp
  - dependencies/include/carla\client\Vehicle.cpp
  - dependencies/include/carla\client\Walker.cpp
  - dependencies/include/carla\client\WalkerAIController.cpp
  - dependencies/include/carla\client\Waypoint.cpp
  - dependencies/include/carla\client\World.cpp
  - dependencies/include/carla\client\detail\ActorFactory.cpp
  - dependencies/include/carla\client\detail\ActorState.cpp
  - dependencies/include/carla\client\detail\ActorVariant.cpp
  - dependencies/include/carla\client\detail\Client.cpp
  - dependencies/include/carla\client\detail\Episode.cpp
  - dependencies/include/carla\client\detail\EpisodeProxy.cpp
  - dependencies/include/carla\client\detail\EpisodeState.cpp
  - dependencies/include/carla\client\detail\Simulator.cpp
  - dependencies/include/carla\client\detail\WalkerNavigation.cpp
  - dependencies/include/carla\geom\GeoLocation.cpp
  - dependencies/include/carla\geom\Math.cpp
  - dependencies/include/carla\geom\Mesh.cpp
  - dependencies/include/carla\nav\Navigation.cpp
  - dependencies/include/carla\nav\WalkerEvent.cpp
  - dependencies/include/carla\nav\WalkerManager.cpp
  - dependencies/include/carla\opendrive\OpenDriveParser.cpp
  - dependencies/include/carla\opendrive\parser\ControllerParser.cpp
  - dependencies/include/carla\opendrive\parser\GeometryParser.cpp
  - dependencies/include/carla\opendrive\parser\GeoReferenceParser.cpp
  - dependencies/include/carla\opendrive\parser\JunctionParser.cpp
  - dependencies/include/carla\opendrive\parser\LaneParser.cpp
  - dependencies/include/carla\opendrive\parser\ObjectParser.cpp
  - dependencies/include/carla\opendrive\parser\ProfilesParser.cpp
  - dependencies/include/carla\opendrive\parser\RoadParser.cpp
  - dependencies/include/carla\opendrive\parser\SignalParser.cpp
  - dependencies/include/carla\opendrive\parser\TrafficGroupParser.cpp
  - dependencies/include/carla\road\Lane.cpp
  - dependencies/include/carla\road\LaneSection.cpp
[... 46 lines cut ...]
running egg_info
writing source\carla.egg-info\PKG-INFO
writing dependency_links to source\carla.egg-info\dependency_links.txt
writing top-level names to source\carla.egg-info\top_level.txt
reading manifest file 'source\carla.egg-info\SOURCES.txt'
writing manifest file 'source\carla.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
running install_lib
running build_py
running build_ext
building 'carla.libcarla' extension
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Idependencies/include "-IC:\Program Files\Python38\include" "-IC:\Program Files\Python38\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\cppwinrt" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.183 [... cut ...]
C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt\stdlib.h(1015): note: see declaration of 'wcstombs'
f:\programs\project\carla\pythonapi\carla\source\libcarla\OSM2ODR.cpp(7): fatal error C1083: Cannot open include file: 'OSM2ODR.h': No such file or directory
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Community\\VC\\Tools\\MSVC\\14.16.27023\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "F:\Programs\Project\Carla\PythonAPI\carla\dist"!

Axel1092, 2020-09-29:
Hi @H-Whisky, sorry for the delay, I need the error log of the whole console, you only sent the PythonAPI part but what is actually failing is one of the dependencies. I need you to delete everything inside the Build folder, run `make osm2odr` then send the full error log in order to better understand what your problem is and solve it.

mc275, 2020-10-09:
> Hi @H-Whisky, sorry for the delay, I need the error log of the whole console, you only sent the PythonAPI part but what is actually failing is one of the dependencies. I need you to delete [...]

I also meet the same problem. I make osm2odr and found the xerces cannot output the lib. When I build the xerces lib, I found  Windows kit 10 already have the icucc.lib and xerces cannot link. Could you please help me to solve this problem?

Axel1092, 2020-10-13:
@mc275 So you mean that that library is missing or it already exists? Can you share the full console log as well please?

MrMojitoo, 2020-10-25:
Hello there,

Having the same issue here, I did what you said @Axel1092 

> Hi @H-Whisky, sorry for the delay, I need the error log of the whole console, you only sent the PythonAPI part but what is actually failing is one of the dependencies. I need you to delete [...]

and I got this (sorry for it being in french) : 
`C:\CARLA\carla>make osm2odr
-[BuildOSM2ODR]: [Batch params]: --build
-- Setting build type to 'Release' as none was specified.
-- Selecting Windows SDK version 10.0.17763.0 to target Windows 10.0.18363.
-- The C compiler identification is MSVC 19.16.27043.0
-- The CXX compiler identification is MSVC 19.16.27043.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CMAKE_BINARY_DIR: C:/CARLA/carla/Build/osm2odr-visualstudio
-- CMAKE_SOURCE_DIR: C:/CARLA/carla/Util/OSM2ODR
--
-- Platform:
--     Host: Windows-10.0.18363 AMD64
--     Target: Windows-10.0.18363 AMD64
--     CMake: 3.18.2
--     CMake generator: Visual Studio 15 2017 Win64
--     CMake build tool: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/MSBuild/15.0/Bin/MSBuild.exe
--     Compiler: MSVC 19.16.27043.0
--
-- Found PythonInterp: C:/Users/Nicolas/anaconda3/python.exe (found version "3.8.3")
-- Found Python: C:/Users/Nicolas/anaconda3/python.exe
CMake Error at C:/Users/Nicolas/AppData/Local/Packages/PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0/LocalCache/local-packages/Python38/site-packages/cmake/data/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:165 (message):
  Failed to find XercesC (missing: XercesC_VERSION)
Call Stack (most recent call first):
  C:/Users/Nicolas/AppData/Local/Packages/PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0/LocalCache/local-packages/Python38/site-packages/cmake/data/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:458 (_FPHSA_FAILURE_MESSAGE)
  C:/Users/Nicolas/AppData/Local/Packages/PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0/LocalCache/local-packages/Python38/site-packages/cmake/data/share/cmake-3.18/Modules/FindXercesC.cmake:103 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:86 (find_package)

-- Configuring incomplete, errors occurred!
See also "C:/CARLA/carla/Build/osm2odr-visualstudio/CMakeFiles/CMakeOutput.log".
Microsoft (R) Build Engine version 15.9.21+g9802d43bc3 pour .NET Framework
Copyright (C) Microsoft Corporation. Tous droits réservés.

MSBUILD : error MSB1009: Le fichier projet n'existe pas.
Commutateur : install.vcxproj
-[BuildOSM2ODR]: OSM2ODR has been successfully installed in "C:\CARLA\carla\PythonAPI\carla\dependencies\"`

Does it help ?

AdelinaK, 2020-11-09:
I have the same issue. Did you find a solution?

mc275, 2020-11-10:
> I have the same issue. Did you find a solution?

The reason is that xerces lib conflict with conda and VS2017. I modify the setup.py and set the  xerces lib path

AdelinaK, 2020-11-10:
> > I have the same issue. Did you find a solution?
> 
> The reason is that xerces lib conflict with conda and VS2017. I modify the setup.py and set the xerces lib path

More exactly where? I'm a little confused

Axel1092, 2020-11-11:
Hi all, @MrMojitoo sorry for the late response. The problem is that the library XercesC is missing from your system. Before 0.9.10 for ubuntu, you need to install it in your system with `sudo apt-get libxerces-c-dev`. After 0.9.10, You do not need to install it yourself, our build system will donwload it and compile it automatically.

icerooqiu, 2020-11-30:
It seems like Xerces can't be installed properly, here is one solution I found.
Install Xerces by `conda install -c anaconda xerces-c`
Then find Anaconda3 folder, copy-paste everything from `\Anaconda3\pkgs\xerces-c-3.2.3-ha925a31_0\Library` to `carla\Build\xerces-c-3.2.3-install`. Delete folder `\carla\Build\osm2odr-visualstudio`, then `make PythonAPI`. 
Folder `carla\PythonAPI\carla\dist` should now be available.

brieucdnl, 2021-01-08:
I'm sorry but i'm also having the issue with osm2odr. 

myPath\mingw64\include\zconf.h(475): fatal error C1083: Impossible d'ouvrir le fichier include : 'unistd.h' : No such file or directory (compilation du fichier source C:\Dev\Carla\Util\OSM2ODR\src\utils\iodevices\OutputDevice_File.cpp) [C:\Dev\Carla\Build\osm2odr-visualstudio\src\osm2odr.vcxproj]
myPath\mingw64\include\zconf.h(475): fatal error C1083: Impossible d'ouvrir le fichier include : 'unistd.h' : No such file or directory (compilation du fichier source C:\Dev\Carla\Util\OSM2ODR\src\utils\xml\SUMOSAXReader.cpp) [C:\Dev\Carla\Build\osm2odr-visualstudio\src\osm2odr.vcxproj]
myPath\mingw64\include\zconf.h(475): fatal error C1083: Impossible d'ouvrir le fichier include : 'unistd.h' : No such file or directory (compilation du fichier source C:\Dev\Carla\Util\OSM2ODR\src\utils\options\OptionsIO.cpp) [C:\Dev\Carla\Build\osm2odr-visualstudio\src\osm2odr.vcxproj]

A support on this one would be really appreciated ! What surprises me it's that unistd.h is a header from linux ...

amzoo, 2021-01-22:
> Hi all, @MrMojitoo sorry for the late response. The problem is that the library XercesC is missing from your system. Before 0.9.10 for ubuntu, you need to install it in your system with `sudo [...]

Same Issue on Ubuntu 18.04, Carla 0.9.11
Solution for me:
Run `sudo apt-get libxerces-c-dev`
Delete all xercesc from Build folder `rm -rf ~/carla/Build/xercesc*`
Run `make PythonAPI`

Kakaluoto, 2021-02-15:
Thanks a lot , your comment reall helped me a lot @icerooqiu

Axel1092, 2021-03-03:
@brieucdnl, your issue seems to be related to your visual studio installation as the unistd.h file mentioned in your error log is visual studio one.

H-Whisky, 2021-03-11:
> @brieucdnl, your issue seems to be related to your visual studio installation as the unistd.h file mentioned in your error log is visual studio one.

> Hi @H-Whisky, sorry for the delay, I need the error log of the whole console, you only sent the PythonAPI part but what is actually failing is one of the dependencies. I need you to delete [...]

Sorry to reply you so late, I have already got the answer, thank you for your help.

sivaprasadraju, 2021-05-17:
Hi @H-Whisky Can you please tell me how you solved the issue?

uservinu, 2021-05-21:
On windows 10, I solved this by downloading xerces from the apache website https://mirror.synyx.de/apache//xerces/c/3/sources/xerces-c-3.2.3.zip and extracting the contents to  <D:>\carla\Build\xerces-c-3.2.3-source. Make sure the folder name is xerces-c-3.2.3-source. Then run make PythonAPI

llyyaa, 2021-06-01:
I've solved this problem on windows.
If you are in a same situation with me, open your xercesC source with VS, and you might see some variables in CMAKE config like "ICU_UCONV_EXECUTABLE" have a value pointing to some .exe file in your " \Anaconda3\Library\bin" folder, these libraries caused some conflict.
To solve this, remove any path like "Anaconda3\Library\bin", "Anaconda3\Library\usr\bin", "Anaconda3\Library\mingw-w64\bin" from your "Path" enviroment variable, then restart your computer and open xercesC source with VS again. You should see those CMAKE variables  having suffix "-NOTFOUND". Rerun make PythonAPI and the xercesC should be compiled successfully. And OSM2ODR.h can be generated.

wrencanfly, 2021-12-08:
> It seems like Xerces can't be installed properly, here is one solution I found.
> 
> * Install Xerces by `conda install -c anaconda xerces-c`
> * Then find Anaconda3 folder, copy-paste everything from `Anaconda3\pkgs\xerces-c-3.2.3-ha925a31_0\Library` to `carla\Build\xerces-c-3.2.3-install`.
> * Next, copy-paste file `xerces-c_3.lib` from `carla\Build\xerces-c-3.2.3-install\lib` to folder `carla\PythonAPI\carla\dependencies\lib`
> * Delete folder `carla\Build\osm2odr-visualstudio`, then `make PythonAPI`.
> 
> Folder `carla\PythonAPI\carla\dist` should now be available.

Thanks so much, that's really helpful. I really appreciate it!!!!! I have been stuck on this problem for a long time. lol

ChinaShrimp, 2023-03-03:
For anyone who don't use conda, here is the step to build xerces manually:

1. First delete all xerces* dirs under `Build`;
2. Run following commands in cmd window:
```
cd <carla source root directory>
.\Util\InstallersWin\install_xercesc.bat --build-dir .\Build
```
3. After compiling successfully, copy file `Build\xerces-c-3.2.3-install\lib\xerces-c_3.lib` to directory `PythonAPI\carla\dependencies\lib`. All above directory are relative path to carla source root path.

ouyangkaikai, 2023-03-14:
> It seems like Xerces can't be installed properly, here is one solution I found.
>
> * Install Xerces by `conda install -c anaconda xerces-c`
> * Then find Anaconda3 folder, copy-paste everything from [...]

Thanks so much, that's really helpful.

Hassan-BABAOUSMAIL, 2023-04-19:
> For anyone who don't use conda, here is the step to build xerces manually:
>
> 1. First delete all xerces* dirs under `Build`;
> 2. Run following commands in cmd window:
>
> ```
> cd <carla source root [...]

Thank you so much, your comment was very usefull. It's working now CARLA 0.9.11

AndritMuca, 2023-05-19:
Hi, im having some issues while Building PythonAPI. Could please help. I copy-paste file xerces-c_3.lib from carla\Build\xerces-c-3.2.3-install\lib to folder carla\PythonAPI\carla\dependencies\lib, still have same issue.. Carla -14 version im using with UE for Carla and visual studio 2019.

Error after copy the xerec files and running make PythonAPI again:

-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --all
-[Setup]: Asynchronous jobs:  4
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Install directory:  "C:\Users\andri\source\repos\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "C:\Users\andri\source\repos\carla\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "C:\Users\andri\source\repos\carla\Build\zlib-install\" if you want to force a rebuild.
    -[install_zlib]: Exiting...
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "C:\Users\andri\source\repos\carla\Build\" --zlib-install-dir "C:\Users\andri\source\repos\carla\Build\zlib-install\"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "C:\Users\andri\source\repos\carla\Build\libpng-1.2.37-install\" if you want to force a rebuild.
    -[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "C:\Users\andri\source\repos\carla\Build\"
    -[install_rpclib]: A rpclib installation already exists.
    -[install_rpclib]: Delete "C:\Users\andri\source\repos\carla\Build\rpclib-install\" if you want to force a rebuild.
    -[install_rpclib]: Exiting...
-[Setup]: Installing Google Test...
    -[install_gtest]: [Batch params]: --build-dir "C:\Users\andri\source\repos\carla\Build\"
    -[install_gtest]: A Google Test installation already exists.
    -[install_gtest]: Delete "C:\Users\andri\source\repos\carla\Build\gtest-install\" if you want to force a rebuild.
    -[install_gtest]: Exiting...
-[Setup]: Installing "Recast & Detour"...
    -[install_recast]: [Batch params]: --build-dir "C:\Users\andri\source\repos\carla\Build\"
    -[install_recast]: A "Recast & Detour" installation already exists.
    -[install_recast]: Delete "C:\Users\andri\source\repos\carla\Build\recast-0b13b0-install\" if you want to force a rebuild.
    -[install_recast]: Exiting...
-[Setup]: Installing Boost...
    -[install_boost]: [Batch params]: --build-dir "C:\Users\andri\source\repos\carla\Build\" --toolset msvc-14.2 --version 1.80.0 -j 4
    -[install_boost]: A Boost installation already exists.
    -[install_boost]: Delete "C:\Users\andri\source\repos\carla\Build\boost-1.80.0-install\" if you want to force a rebuild.
    -[install_boost]: Exiting...
-[Setup]: Installing Xercesc...
    -[install_xercesc]: [Batch params]: --build-dir "C:\Users\andri\source\repos\carla\Build\"
    -[install_xercesc]: A xerces installation already exists.
    -[install_xercesc]: Delete "C:\Users\andri\source\repos\carla\Build\xerces-c-3.2.3-install\" if you want to force a rebuild.
    -[install_xercesc]: Exiting...
        1 file(s) copied.
-[Setup]: Installing Sqlite3
    -[install_sqlite3]: [Batch params]: --build-dir "C:\Users\andri\source\repos\carla\Build\"
    -[install_sqlite3]: A Sqlite3 installation already exists.
    -[install_sqlite3]: Delete "C:\Users\andri\source\repos\carla\Build\sqlite3-install" if you want to force a rebuild.
    -[install_sqlite3]: Exiting...
        1 file(s) copied.
-[Setup]: Installing PROJ
    -[install_proj]: [Batch params]: --build-dir "C:\Users\andri\source\repos\carla\Build\"
    -[install_proj]: A PROJ installation already exists.
    -[install_proj]: Delete "C:\Users\andri\source\repos\carla\Build\proj-install" if you want to force a rebuild.
    -[install_proj]: Exiting...
        1 file(s) copied.
-[Setup]: Installing Eigen
    -[install_eigen]: [Batch params]: --build-dir "C:\Users\andri\source\repos\carla\Build\"
0 File(s) copied
0 File(s) copied
        1 file(s) copied.
        1 file(s) copied.

    -[install_eigen]: Eigen has been successfully installed in "C:\Users\andri\source\repos\carla\Build\eigen-install"!
    -[install_eigen]: Exiting...
-[Setup]: Creating "CMakeLists.txt.in"...
-[Setup]:
   ###########
   # SUCCESS #
   ###########
[... 69 lines cut ...]
-- Configuring done (0.3s)
-- Generating done (0.2s)
-- Build files have been written to: C:/Users/andri/source/repos/carla/Build/osm2odr-visualstudio
Microsoft (R) Build Engine version 16.11.2+f32259642 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  osm2odr.vcxproj -> C:\Users\andri\source\repos\carla\Build\osm2odr-visualstudio\src\Release\osm2odr.lib
  -- Install configuration: "Release"
-[BuildOSM2ODR]: OSM2ODR has been successfully installed in "C:\Users\andri\source\repos\carla\PythonAPI\carla\dependencies\"
-[BuildPythonAPI]: [Batch params]: --py3
Building Python API for Python 3.
compiling:
  - source/libcarla/libcarla.cpp
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: invalid command 'bdist_wheel'

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "C:\Users\andri\source\repos\carla\PythonAPI\carla\dist"!

I also have installed and upgrade wheel and seutptools.

Thank you for any help

VidaAPI, 2023-06-14:
Hi @Alex1091 and @germanros1987 I'm having the same issue as @AndritMuca . Info: Carla 14 version and visual studio 2019. I'd appreciate your solutions to how to proceed:

Upon building CARLA Python API I get stuck in this part with the following error:
  -- Install configuration: "Release"
-[BuildLibCarla]: LibCarla server has been successfully installed in "C:\carla\Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies\"!
-[BuildLibCarla]: LibCarla client has been successfully installed in "C:\carla\PythonAPI\carla\dependencies\"!
-[BuildOSM2ODR]: [Batch params]: --build --all
HEAD is now at ee0c2b9241f Removed debug warnings
-- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19045.
-- CMAKE_BINARY_DIR: C:/carla/Build/osm2odr-visualstudio
-- CMAKE_SOURCE_DIR: C:/carla/Build/om2odr-source
--
-- Platform:
--     Host: Windows-10.0.19045 AMD64
--     Target: Windows-10.0.19045 AMD64
--     CMake: 3.25.3
--     CMake generator: Visual Studio 16 2019
--     CMake build tool: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/MSBuild/Current/Bin/MSBuild.exe
--     Compiler: MSVC 19.29.30148.0
--
CMake Error at C:/Program Files/CMake/share/cmake-3.25/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Failed to find XercesC (missing: XercesC_VERSION)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.25/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  C:/Program Files/CMake/share/cmake-3.25/Modules/FindXercesC.cmake:112 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:71 (find_package)

 

-- Configuring incomplete, errors occurred!
See also "C:/carla/Build/osm2odr-visualstudio/CMakeFiles/CMakeOutput.log".
Microsoft (R) Build Engine version 16.11.2+f32259642 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

 

MSBUILD : error MSB1009: Project file does not exist.
Switch: install.vcxproj
-[BuildOSM2ODR]: OSM2ODR has been successfully installed in "C:\carla\PythonAPI\carla\dependencies\"
-[BuildPythonAPI]: [Batch params]: --py3
Building Python API for Python 3.
compiling:
  - source/libcarla/libcarla.cpp
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

 

error: invalid command 'bdist_wheel'

emlachapelle, 2023-07-26:
> For anyone who don't use conda, here is the step to build xerces manually:
>
> 1. First delete all xerces* dirs under `Build`;
> 2. Run following commands in cmd window:
>
> ```
> cd <carla source root [...]

I have tried this multiple times but I am getting an error... I am on windows 10, trying to install carla 9.13 from source... any help would be greatly appreciated!

vicmen14, 2023-11-25:
> Hi, I'm building on Windows and make launch runs fine. When I run make PythonAPI, I receive this error log. it finishes with:
>
> -[BuildPythonAPI]: Carla lib for python has been successfully installed [...]

It is due to an error on Setup.bat file. It tries to copy xerces lib to a non existing path. Change this:

copy %INSTALLATION_DIR%\xerces-c-3.2.3-install\lib\xerces-c_3.lib %CARLA_PYTHON_DEPENDENCIES%\lib
copy %INSTALLATION_DIR%\xerces-c-3.2.3-install\lib\xerces-c_3.lib %CARLA_DEPENDENCIES_FOLDER%\lib

to

xcopy /Y /S /I "%INSTALLATION_DIR%\xerces-c-3.2.3-install\lib\xerces-c_3.lib" "%CARLA_PYTHON_DEPENDENCIES%\lib" > NUL
xcopy /Y /S /I "%INSTALLATION_DIR%\xerces-c-3.2.3-install\lib\xerces-c_3.lib" "%CARLA_DEPENDENCIES_FOLDER%\lib" > NUL

do the same for sqlite3

CMakey, 2023-11-30:
> It seems like Xerces can't be installed properly, here is one solution I found.
>
> * Install Xerces by `conda install -c anaconda xerces-c`
> * Then find Anaconda3 folder, copy-paste everything from [...]

it seems doesn't work today, after following these steps,error still says '“OSM2ODR.h”: No such file or directory'.
while i'm using win11 and python 3.12, is there any possible that the python version is the problem?

CMakey, 2023-11-30:
#6898 
anyone who get error like 'CMake Error: The source directory "E:/GitHub/carla/Build/osm2odr-visualstudio/x64" does not exist.' and find there is nothing in u folder 'E:\GitHub\carla\Build\osm2odr-visualstudio'is empty . 

and u had tried to reinstall  xerces but nothing worked , u can tried to change the `.\Util\BuildTools\BuildOSM2ODR.bat` .

it works for me!

CMakey, 2023-11-30:
after generated the `dist` folder and wheel,egg files, i find is came errors that i can't install the wheel, which report 'platform' error.
i saw some one solve this by downgrade python , i did the same thing . But when I tried to use `make PythonAPI` to regenerate wheel. it came error `error: invalid command 'bdist_wheel'`. and i don't find any sulotion.

pseventekidis, 2023-12-05:
I am having also the same issue. Maybe I will try the Linux build

pseventekidis, 2023-12-05:
> after generated the `dist` folder and wheel,egg files, i find is came errors that i can't install the wheel, which report 'platform' error. i saw some one solve this by downgrade python , i did the [...]

Actually try this
https://github.com/carla-simulator/carla/issues/6898#issuecomment-1807921422

pseventekidis, 2023-12-05:
> #6898 anyone who get error like 'CMake Error: The source directory "E:/GitHub/carla/Build/osm2odr-visualstudio/x64" does not exist.' and find there is nothing in u folder [...]

try this also 
https://github.com/carla-simulator/carla/issues/6898#issuecomment-1807921422

kuanting, 2024-01-21:
> it seems doesn't work today, after following these steps,error still says '“OSM2ODR.h”: No such file or directory'. while i'm using win11 and python 3.12, is there any possible that the python [...]

I have encountered the same issue. The `xerces-c_3.lib` copy-paste method didn't work!

thibaud-siamer, 2024-01-22:
Hello,
Removing quotes around %OSM2ODR_SOURCE_PATH% in carla\Util\BuildTools\BuildOSM2ODR.bat line 119 seems to work.

shangwp, 2024-01-25:
> Hello, 你好，我是 Removing quotes around %OSM2ODR_SOURCE_PATH% in carla\Util\BuildTools\BuildOSM2ODR.bat line 119 seems to work. 删除carla\Util\BuildTools\BuildOSM2ODR.bat第119行中%OSM2ODR_Source_PATH %周围的引号似乎可以工作。

It works for me, win11 + carla9.15, thanks!

mhvicom, 2024-01-31:
> It seems like Xerces can't be installed properly, here is one solution I found.
>
>     * Install Xerces by `conda install -c anaconda xerces-c`
>
>     * Then find Anaconda3 folder, copy-paste everything [...]

In case someone finds this useful, combining this fix and [this one](https://github.com/carla-simulator/carla/issues/6898#issuecomment-1807921422) have worked for me.

pseventekidis, 2024-01-31:
I had the same issue today. 
It could be from bad internet connection.
Consider the above comments but also try to erase the build directory and try again with better internet. 
It worked for me

AnaStasiaS-S, 2024-02-08:
> 1. xerces*

You are genius! Thank you a lot!!!
Works on Win11, carla 0.19.5

AnaStasiaS-S, 2024-02-08:
> Hello, Removing quotes around %OSM2ODR_SOURCE_PATH% in carla\Util\BuildTools\BuildOSM2ODR.bat line 119 seems to work.

You are genius! Thank you a lot!!!
Works on Win11, carla 0.19.5

BrianLiu2005, 2024-02-11:
> Hello, Removing quotes around %OSM2ODR_SOURCE_PATH% in carla\Util\BuildTools\BuildOSM2ODR.bat line 119 seems to work.

This also worked for me too (Windows 10, Carla 0.19.5), thanks a bunch!

LarsWeber123, 2024-02-20:
I currently experience the same problem (Can ot open OSM2ODR.H) at a different location:
My setting is Windows 10, Visual Studio2022, Carla 0.9.15, UE4.26

I call make PythonAPI GENERATOR="Visual Studio 17 2022"
 and until this point "BuildLibCarla" it works fine, but it fails when creating osm2odr.

```

-[BuildLibCarla]: LibCarla server has been successfully installed in "c:\Carla\carla\Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies\"!
-[BuildLibCarla]: LibCarla client has been successfully installed in "c:\Carla\carla\PythonAPI\carla\dependencies\"!
-[BuildOSM2ODR]: [Batch params]: --generator "Visual Studio 17 2022" --build --all
HEAD is now at 1835e1e9538 Remove logs
CMake Warning:
  Ignoring extra path from command line:

   "c:\Carla\carla\Build\om2odr-source""

CMake Error: The source directory "C:/Carla/carla/Build/osm2odr-visualstudio/x64" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
Error: could not load cache
Das System kann die angegebene Datei nicht finden.
Das System kann die angegebene Datei nicht finden.
-[BuildOSM2ODR]: OSM2ODR has been successfully installed in "c:\Carla\carla\PythonAPI\carla\dependencies\"
-[BuildPythonAPI]: [Batch params]: --py3
Building Python API for Python 3.
compiling:
  - source/libcarla/libcarla.cpp
running bdist_egg
running egg_info
writing source\carla.egg-info\PKG-INFO
writing dependency_links to source\carla.egg-info\dependency_links.txt
writing top-level names to source\carla.egg-info\top_level.txt
reading manifest file 'source\carla.egg-info\SOURCES.txt'
writing manifest file 'source\carla.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
C:\python\Lib\site-packages\setuptools\command\install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
running install_lib
running build_py
running build_ext
building 'carla.libcarla' extension
"C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.38.33130\bin\HostX86\x64\cl.exe" /c /nologo /O2 /W3 /GL /DNDEBUG /MD -Idependencies/include -IC:\python\include -IC:\python\Include "-IC:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.38.33130\include" "-IC:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.38.33130\ATLMFC\include" "-IC:\Program Files\Microsoft Visual Studio\2022\Community\VC\Auxiliary\VS\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22621.0\\um" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22621.0\\shared" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22621.0\\winrt" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22621.0\\cppwinrt" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.38.33130\include" "-IC:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.38.33130\ATLMFC\include" "-IC:\Program Files\Microsoft Visual Studio\2022\Community\VC\Auxiliary\VS\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22621.0\\um" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22621.0\\shared" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22621.0\\winrt" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22621.0\\cppwinrt" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" /EHsc /Tpsource/libcarla/libcarla.cpp /Fobuild\temp.win-amd64-cpython-311\Release\source/libcarla/libcarla.obj /experimental:external /external:W0 /external:I dependencies/include/system /DBOOST_ALL_NO_LIB /DBOOST_PYTHON_STATIC_LIB /DBOOST_ERROR_CODE_HEADER_ONLY /D_WIN32_WINNT=0x0600 /DHAVE_SNPRINTF /DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true /MD
libcarla.cpp
c:\Carla\carla\PythonAPI\carla\source\libcarla\OSM2ODR.cpp(7): fatal error C1083: Datei (Include) kann nicht geöffnet werden: "OSM2ODR.h": No such file or directory
error: command 'C:\\Program Files\\Microsoft Visual Studio\\2022\\Community\\VC\\Tools\\MSVC\\14.38.33130\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "c:\Carla\carla\PythonAPI\carla\dist"!
```

The problem starts with the missing x64 folder and ends with a missing header file requested "c:\Carla\carla\PythonAPI\carla\source\libcarla\OSM2ODR.cpp(7): fatal error C1083: Datei (Include) kann nicht geöffnet werden: "OSM2ODR.h": No such file or directory"

I can find this header in "C:\Carla\carla\Build\om2odr-source\src" but this path is not in the include statements -I for the cl.exe call above. Maybe it should have been copied somewhere else, probably into the missing x64 folder.

I also have a bit of trouble understanding how the previous step of building the Unreal Engine is linked to this "building Carla" part. Can this problem actually have something to do with the previous step of compiling the Engine? Although that went fine, I can open the UE editor and it seems to work as expected.

closed 2024-11-06 by Blyron

sleepyLeo6, 2025-01-05:
> Hello, Removing quotes around %OSM2ODR_SOURCE_PATH% in carla\Util\BuildTools\BuildOSM2ODR.bat line 119 seems to work.

It works for me with win11 + carla 0.9.15 + vs2019.