carla-simulator/carla #3324: what is the lidar and IMU coordinate direction?

Maxinjun, 2020-09-20:
carla version:0.9.9 +Ubuntu16.04

it seem that lidar coordinate direction(x-right, y-backward, z-down)
anyOne know IMU coordinate direction?

germanros1987, 2020-09-24:
@DSantosO for an accurate explanation.
@sergi-e to make sure this is clear in the documentation

DSantosO, 2020-09-24:
Hello @Maxinjun,

In CARLA we use the Unreal system of reference that is:  X: Foward,  Y-Right, Z: Up
so both Lidar and IMU follow this coordinate system. Notice that both are given in the local coordinate system of the sensor, not in the global one.
Be careful with this coordinate system because most of the visualization software uses the inverse convention for the Y-axis so you will see everything mirror in this axis. If you check our script that visualizes the lidar with Open3D, we are flipping this coordinate for consistency with the Carla frame.
For the IMU, the accelerometer uses this coordinate frame, if you check the manual_control script, you will see that the vehicle forwardly accelerating results in x-acceleration measurement. Respect to the gyroscope, the same is true, measuring the rotation respect to X, Y, and Z defined as I commented above.

sergi-e, 2020-10-01:
Added a note regarding coordinate system and sensors in the sensors reference!

closed 2020-10-15 by sergi-e

lsangreg, 2023-12-13:
And which Reference system are the camera following? The rotation  quaternion from ego_vehicle to a camera positioned in front roll,pitch,yaw = [0,0,0] is: -0.5 0.5 -0.5 0.5. 
It means that for example we can have a rotation like this [ x: -1.5707963, y: 0, z: -1.5707963 ] with euler angles ZXY. 
So we have z up, x left,and y entering the plane. 
Is this the RS of the cameras?