carla-simulator/carla #3336: Attribute error while running scenario_runner.py in Carla 0.9.8

NishanthShettyS, 2020-09-22:
Hi,
I am running carla 0.9.8 on Windows 10.
Carla simulator is working fine and I was able to run the example files given in PythonAPI folder.
But when I tried to run the scenarios from scenario runner (0.9.8 version), I faced the below error 

`File D:\Carla\CARLA_0.9.8\scenario_runner-0.9.8\srunner\scenariomanager\scenarioatomics\atomic_behaviors.py", line 1572, in TrafficLightManipulator
    RED = carla.TrafficLightState.Red
AttributeError: module 'carla' has no attribute 'TrafficLightState'`

I had followed the instruction given in "getting started.md" file and had set the Python paths as given before running the scenario. 
Could you please tell me how to solve this ? Thank you!

germanros1987, 2020-09-24:
Is this still an issue in 0.9.9? @Axel1092 could you check this?

glopezdiest, 2020-09-30:
Hey @NishanthShettyS. Seems like scenario runner can't retrieve the location of CARLA's PythonAPI. Could you share your PythonPath? you should have something like *${CARLA_ROOT}/PythonAPI* there

closed 2020-12-25 by stale[bot]