carla-simulator/carla #3337: Error converting OpenStreetMap to Opendrive

agourishetty93, 2020-09-22:
Issue while converting the OpenstreetMap .osm file to Opendrive format

Generated a map using openstreetmap.org and saved as map.osm and used the following code to convert to Opendrive format as mentioned in the document. https://carla.readthedocs.io/en/latest/tuto_G_openstreetmap/

>>>>
import carla

client = carla.Client('127.0.0.1', 2000)
client.set_timeout(10.0)

# Read the .osm data
osm_file = open("............\\Co-Simulation\\Sumo\\map.osm", 'r')
osm_data = osm_file.read()
osm_file.close()

# Define the desired settings. In this case, default values.
settings = carla.Osm2OdrSettings()
# Convert to .xodr
xodr_data = carla.Osm2Odr.convert(osm_data, settings)

# save opendrive file
f = open("....................\\PythonAPI\\util\\opendrive\\map.xodr", 'w')
f.write(xodr_data)
f.close()

#**i am using python 3.6 and it throws the following error** 

Traceback (most recent call last):
  File "...........................

a/PythonAPI/examples/Convert_to_opendriveformat.py", line 10, in <module>
    osm_data = osm_file.read()
  File "............\Python36\lib\encodings\cp1252.py", line 23, in decode
    return codecs.charmap_decode(input,self.errors,decoding_table)[0]
UnicodeDecodeError: 'charmap' codec can't decode byte 0x81 in position 71102: character maps to <undefined>

attached the codecs code of cp1252.py
[image]

germanros1987, 2020-09-24:
@Axel1092 could you check that this works before the release?

Axel1092, 2020-09-24:
Hi @agourishetty93, your issue is related to character encoding. OSM data is usually written in UTF-8 which in windows you need to specify it. In your script you should change 
`osm_file = open("............\Co-Simulation\Sumo\map.osm", 'r')` 
to 
`osm_file = open("............\Co-Simulation\Sumo\map.osm", 'r', encoding='utf-8')`

nagasanjay, 2020-10-10:
I had the same error while running with `config.py` .
I even added `encoding='utf-8` at `open()` function. But didn't help

agourishetty93, 2020-10-12:
HI Axel, the issue was resolved after updating the carla version to 0.9.10 and with the character encoding changes. Thankyou BR

closed 2020-12-25 by stale[bot]