carla-simulator/carla #3338: Docker build fails to "ran out of memory"

lisahabich, 2020-09-22:
Hi, we are trying to build the docker image to be able to import new maps, but the build process fails.

Error at "docker build -t carla -f Carla.Dockerfile .":

 LogMemory: Warning: MemoryStats:
  	AvailablePhysical 593063936
  	 AvailableVirtual 324964352
  	     UsedPhysical 3760545792
  	 PeakUsedPhysical 4385202176
  	      UsedVirtual 6799183872
  	  PeakUsedVirtual 7461810176
  LogCore: Error: appError called: Fatal error: [File:/home/ue4/UnrealEngine/Engine/Source/Runtime/Core/Private/Unix/UnixPlatformMemory.cpp] [Line: 1027]
  Ran out of memory allocating 1073807360 bytes with alignment 65536

**System information:**
- Ubuntu 18.4
- 12 GB memory
- 350 GB storage

We tried several amounts of memory, but no change at all.
tia

germanros1987, 2020-09-24:
@bernatx have you seen this error? Do you have any recommendations?

bernatx, 2020-09-24:
I have not seen this error on RAM memory, but I think you need almost 700 Gb free of HD to build the Docker, as it is very expensive in HD space when building.

closed 2020-12-25 by stale[bot]