carla-simulator/carla #3340: Testing (Autoexposure & Lights intensities)

Madecu, 2020-09-22:
We have changed the default exposure mode to autoexposure (Eye adapting).
We also have changed the values of streetlights, car lights, and buildings.

We have to drive around all the maps, changing between day and night, to see if there are some lights that are not lighting properly (too bright or too low). Including all the vehicle lights (breaks, reverse, high beam, blinkers, etc.).

While you are testing that, look out for some artifacts or weird behaviors from the autoexposure. Just in case.

Axel1092, 2020-09-23:
Lightning seems ok and autoexposure works well, most areas are properly lit except when there are no lights (which is ok). Only  in Town03 this area seems too dark but this might be intentional.
[image]

Madecu, 2020-09-24:
- Fixed weird artifacts with the clouds (Turn blue or turn the sky into pitch black)

- Tweaked "Fountain", "FoodCar", "CarlaCola" and "BusStop" light intensities.

- Lowered Skyscraper_04 light intensity

- Changed mustang back blinkers

- Increased spotlight radius in streetlights (simulates global lighting). Now they light further

closed 2020-09-25 by Madecu