carla-simulator/carla #3341: MapBuilder: Lane change default value handling not as in open drive spec

hofbi, 2020-09-22:
There is a difference in the default value of the [MapBuilder](https://github.com/carla-simulator/carla/blob/dev/LibCarla/source/carla/road/MapBuilder.cpp#L198) and the [OpenDrive Spec p. 59](http://www.opendrive.org/docs/OpenDRIVEFormatSpecRev1.4H.pdf) which says for lane chane 

> allow a lane change in the indicated direction taking into account that lanes are numbered in ascending order from right to left. If the attributeis missing, “both” is assumedto be valid.

**Current behavior**

MapBuilder defaults to `none`

**Expected OpenDrive behavior**

MapBuilder defaults to `both`

germanros1987, 2020-09-24:
@hofbi which version of CARLA are you using?

germanros1987, 2020-09-24:
@glopezdiest stay alert for a response of the user

hofbi, 2020-09-24:
The mismatch is for the latest master.

Axel1092, 2020-09-24:
Hi @hofbi, thank you for pointing this discrepancy with OpenDRIVE standard. We have opened a pull request to address this issue.

PR #3361 (2020-11-25): Changed the default value of lane change record
#### Description

OpenDRIVE specification (http://www.opendrive.org/docs/OpenDRIVEFormatSpecRev1.4H.pdf pg 59) requires the lane change record to be defaulted to "both" instead of "none".

Fixes #3341

#### Where has this been tested?

  * **Platform(s):** Ubuntu 18.04
  * **Python version(s):** 3
  * **Unreal Engine version(s):** 4.24

#### Possible Drawbacks
This PR needs to be tested to be sure that it does not break our towns.

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/3361)
closed 2020-11-25 by marcgpuig