carla-simulator/carla #3346: Differ lane types by colors

italovinicius18, 2020-09-22:
I'm trying to mine some data to run a lanenet model on CARLA images, but i have to differ lanes types like broken and solid lines. I already used semantic and rgb camera sensor to get images and i filter the image just to focus on lines, but how i can i differ the color the types of lanes ? I need to use waypoint and waypoint.lane_type ? But how i can apply these to differ the color lanes in image to generate the ground truth data ?

**Raw image:** [image]

**Filtered image:** [image]  

I want that broken lines have a diferent color,it's that but i dont know how to do that

germanros1987, 2020-09-24:
@marcgpuig what would be the simples method to achieve this?

Madecu, 2020-09-29:
Hi @italovinicius18, as far as I know, unfortunately, this is a hard task due to the way RoadRunner gave us the geometry of the lines and roads. To do so, you would need to export all the road lines meshes from unreal and open them with some 3D software such as Maya or 3DsMax. Then, you need to detach the solid lines from the broken ones and import them again into unreal engine. Once you have separated both types of lines in their folders, you have to put them inside the scene at the same location as before. Finally, you have to edit open3d_lidar.py and add the new semantic segmentation categories and assign them a color.

italovinicius18, 2020-10-10:
Thank for the aswer @Madecu

closed 2020-12-25 by stale[bot]