carla-simulator/carla #3347: CARLA hangs in SUMO co-simulation

aniolm9, 2020-09-23:
When running a co-simulation with SUMO and [Artery](https://github.com/riebl/artery), CARLA just hangs and the operating system suggests me to "Force quit". The commands I run are the following:

**Start SUMO server:**
```
sumo --remote-port 8813 --num-clients 2 -c examples/Town01.sumocfg
```

**Start CARLA and set map:**
```
bin/CarlaUE4.sh
python /opt/carla-simulator/PythonAPI/util/config.py --map Town01
```

**Start Omnet++ with Artery:**
```
 ../../build/run_artery.sh sumo-omnetpp.ini
```

**Run synchronization:**
```
python run_synchronization.py examples/Town01.sumocfg --sumo-port 8813 --sumo-host 127.0.0.1 --tls-manager carla
```

Finally, I click `Run` in the OmnetIDE and Carla hangs.

It is important to observe that running the co-simulation just with SUMO and Carla works as expected.

**OS:** Ubuntu 20.04 (5.4.0-48)
**Carla version:** 0.9.9.2-1
**Carla installation source:** Carla Ubuntu repository.
**SUMO version:** 1.7.0
**SUMO installation source:** SUMO PPA.

agourishetty93, 2020-09-23:
Even i have the same issue in windows. https://github.com/carla-simulator/carla/issues/3337

germanros1987, 2020-09-24:
@joel-mb could you take a look?

joel-mb, 2020-11-06:
Hi @aniolm9 ,
Sorry for the late answer, I didn't have the time to look into this. Are you still facing the issue? Do you have any hint what can be happening? I'm replicating your environment to reproduce the issue.

Llagus, 2020-11-18:
Hi @joel-mb,
I'm taking the lead from @aniolm9. The issue is still there and I don't know what can be happening.

joel-mb, 2020-11-23:
Hi @Llagus! where can I find `sumo-omnetpp.ini`? Could you provide me that file? Thanks!

joel-mb, 2020-11-23:
Hi @Llagus,
Can you test adding manually a `projParameter` to the `location` tag of the Town01.net.xml.
Just substitue:
```xml
<location netOffset="0.06,328.61" convBoundary="0.00,0.00,394.44,328.66" origBoundary="-0.06,-328.61,394.38,0.05" projParameter="!"/>
```
by:
```xml

joel-mb, 2020-11-23:
Hi @Llagus,
Can you test adding a `projParameter` to the `location` tag of the Town01.net.xml. Just substitute:
```xml
<location netOffset="0.06,328.61" convBoundary="0.00,0.00,394.44,328.66" origBoundary="-0.06,-328.61,394.38,0.05" projParameter="!"/>
```
with:
```xml
<location netOffset="0.06,328.61" convBoundary="0.00,0.00,394.44,328.66" origBoundary="-0.06,-328.61,394.38,0.05" projParameter="+proj=tmerc +lat_0=0 +lon_0=0 +k=1 +x_0=0 +y_0=0 +datum=WGS84 +units=m +geoidgrids=egm96_15.gtx +vunits=m +no_defs"/>
```
It seems that an error occurs when the SUMO network is not georeferenced.

Llagus, 2020-11-27:
Hi @joel-mb, 
I tried changing the Town01.net.xml file and Carla is still hanging.  

I attach here the sumo-omnetpp.ini file.

[sumo-omnetpp.ini.txt](https://github.com/carla-simulator/carla/files/5607648/sumo-omnetpp.ini.txt)

joel-mb, 2020-12-04:
Hi @Llagus,
After some testing, I've been able to run a co-simulation with CARLA, SUMO and Artery. I'm building artery using the provided Dockerfile in their repository. I also had to introduce minor modifications to the `sumo-omentpp.ini` file in order to make it work. Please, find attached the modified files here:
[artery.zip](https://github.com/carla-simulator/carla/files/5644530/artery.zip)
Let me know if this work for you.

As far as I've seen the problem was that the SUMO client of artery was not ticking the server and therefore CARLA was hanging. Is Artery displaying any message when you press play? Is it running normally or it also freezes? Also, can you try if your simulation it's working without using CARLA? 

I'm using the following versions:
**CARLA** 0.9.10.1
**Artery**: current master
**Omnet**: 5.6.2
**SUMO**: 1.6.0

Llagus, 2020-12-18:
Hi @joel-mb , 

I tried to run the simulation with the files you send me, but Carla is still hanging. I tried to run the simulations independently and the Artery simulation was running fine. 

So after that, I tried to run the co-simulation again and I noticed that Carla is running the simulation as well. When you choose the option of maximum speed simulation in the Artery you can see the cars moving in the Carla as well. But between the steps that Carla does, when is waiting for Artery to finish his step, Carla hangs and you don't have control of it.

TraCi waits for all the clients to send him the order  Simulation Step before it sends the next step to all the clients. Carla does this step really fast but Artery does it slower, and Carla hangs waiting there. I noticed as well that in the short moment Carla  is working you can move the camera, but while waiting the next step the application is frozen.  When the simulation stops Carla works as expected because the connection with TraCi ends. 

I tried this with the files that I have at the beginning and the same occurs.

closed 2021-07-21 by stale[bot]