carla-simulator/carla #3348: object parser non compiling (wrong type cast)

amastrobera, 2020-09-23:
https://github.com/carla-simulator/carla/blob/f2c3b9fce6f1dbba1852996c546fb2127de93371/LibCarla/source/carla/opendrive/parser/ObjectParser.cpp#L73

should be `node_object.attribute("id").as_uint()` not `as_string()`

germanros1987, 2020-09-24:
@amastrobera please provide more context. Is this master, dev, which branch?
@Axel1092 could you take a look?

Axel1092, 2020-09-24:
Hi @amastrobera, it is parsed as string as the OpenDRIVE specifies that the object id is string type and not `uint` and only requires it to be unique: 
[image]

amastrobera, 2020-09-24:
@germanros1987 it's in the master
@Axel1092 you declared the function AddSignal this way. It's expecting a SignId (alias for uint_32t). So, it's not a string and doesn't compile

```
    element::RoadInfoSignal* AddSignalReference(
        Road* road,
        const SignId signal_id,
        const double s_position,
        const double t_position,
        const std::string signal_reference_orientation);
```

https://github.com/carla-simulator/carla/blob/f2c3b9fce6f1dbba1852996c546fb2127de93371/LibCarla/source/carla/road/MapBuilder.h#L154

Axel1092, 2020-09-29:
@amastrobera `SignId` is defined as string (see `RoadTypes.h`)
```
  using RoadId = uint32_t;

  using JuncId = int32_t;

  using LaneId = int32_t;

  using SectionId = uint32_t;

  using ObjId = uint32_t;

  using SignId = std::string;

  using ConId = uint32_t;

  using ContId = std::string;
```

amastrobera, 2020-09-29:
Apologies, my repo must have lost track of the updates on the RoadType.h file. You did it in February.

closed 2020-09-29 by amastrobera