carla-simulator/carla #3358: CarlaUE4.exe can not apply two pygame windows

CHDCheng, 2020-09-24:
Hello, I use Carla 0.9.10, I run CarlaUE4.exe as a server in local pc, and I run the manual_control.py, a pygame window(1280*720) opens and I can control the vehicle, but then I run another manual_control.py at the same time,  an error happend, CarlaUE4.exe has crashed and will close, the content is :

> "Assertion failed: InRect.Height() <= TextureRHI->GetSizeXYZ().Y >> InFlags.GetMip()
> [File:D:/Build/++UE4/Sync/Engine/Source/Runtime/Windows/D3D11RHI/Private/D3D11RenderTarget.cpp][line:506]
> Provided rect height (1440), must be smaller or equal to the texture size requested Mip (720)".

Whatever I run the two manual_control.py in local pc or other pc throuth IP+port, the CarlaUE4.exe crashed everytime.

Then, I opened CarlaUE4.uproject in UE4Editor, and play(not start) the project, I found I could run two manual_control.py, and can apply two pygame windows at the same time.

So, I want to ask that, if there are some ways to solve this problem. Thank you very much!!

CHDCheng, 2020-09-27:
Could you please help me? Thank you very much! @germanros1987

BAROBA-3d, 2020-12-10:
Hi, can I ask how you were able to control Carla? Where do I have to type the commands? I can not find any info about this...
Thanks for any help.

germanros1987, 2021-01-26:
@corkyw10 could you follow up on this?

corkyw10, 2021-01-29:
@CHDCheng Are you still experiencing this issue? I have tested in 0.9.10 and 0.9.11 and I'm not experiencing the same problem, both manual_control.py scripts run fine. Are you running the scripts with any previous changes in configurations or just out of the box?

@BAROBA-3d I think you know CARLA more now, but I will share the [getting started with CARLA docs](https://carla.readthedocs.io/en/latest/core_world/) here for anyone that comes across this in the future.

closed 2021-08-05 by corkyw10