carla-simulator/carla #3366: Error in importing OSM input into road network

Shiyunyang-seu, 2020-09-24:
Hi, I want to use osm as the input to build the network, and I followed the tutorial https://carla.readthedocs.io/en/latest/tuto_G_openstreetmap/ , two different ways were tried but both of them can not be successful:
1. Convert the osm to opendrive and load in carla, but an error occurred: RuntimeError: opendrive could not be correctly parsed
2. Read OSM map directly like: python config.py --osm-path='XXX', but an fatal error occurred.
Attached please find my osm and xodr file and the test code.
My version is carla 0.9.10 pre-windows
Thanks in advance.

[carla_osm_test.zip](https://github.com/carla-simulator/carla/files/5277603/carla_osm_test.zip)

jjyyxx, 2020-10-13:
Any progress?

jakubLangr, 2020-10-16:
I'm also curious

jjyyxx, 2020-10-17:
One interesting finding was, according to my test, road name longer than 12 bytes (e.g. 13 english character or 5 chinese character or mixed) led to crash.

germanros1987, 2021-01-26:
@Axel1092 could you please follow up on this?

Axel1092, 2021-01-27:
Hi @Shiyunyang-seu, I tried your file in the latest 0.9.11 and I found no issue. We fixed a few related problems in the latest release. Let us know if you still find more issues.

EVYang1992, 2021-02-07:
Hi there, i use the API _**carla.Osm2Odr**_ to convert _.osm_ file to _.xodr_ file, following the instruction on this [page](https://carla.readthedocs.io/en/latest/tuto_G_openstreetmap/), I successfully got a _.xodr_ file, but each when i run the standalone opendrive mode, the Carla server crashed, the error showed like this: 
[image],

this is the osm raw file:
[map.zip](https://github.com/carla-simulator/carla/files/5938427/map.zip)

Axel1092, 2021-03-03:
Hi @EVYang1992, sorry for the answer delay. What CARLA version are you using and what settings for carla.Osm2OdrSettings? I have tried with 0.9.11 and your map seems to work properly.

EVYang1992, 2021-04-11:
Hi @Axel1092 , I used 0.9.11，but the fatal always showed up, since you can make it happen to run it, I am wondering if I should make some changes on my laptop. I can pick some other osm files and generate the map in Carla environment, but for the one i uploaded, no i can not generate it other than got that fatal.

Axel1092, 2021-04-13:
Maybe this is a windows issue. I'll check this as soon as possible.

closed 2023-06-02 by stale[bot]