carla-simulator/carla #3374: Debian CARLA on Ubuntu 20.04 error

shri-can, 2020-09-29:
I'm having a problem with the first step of the Debian CARLA installation in Quick Start Instructions.

> $ sudo add-apt-repository "deb [arch=amd64] http://dist.carla.org/carla $(lsb_release -sc) main"
Ign:1 cdrom://Ubuntu 20.04.1 LTS _Focal Fossa_ - Release amd64 (20200731) focal InRelease
Hit:2 cdrom://Ubuntu 20.04.1 LTS _Focal Fossa_ - Release amd64 (20200731) focal Release
Hit:4 http://dl.google.com/linux/chrome/deb stable InRelease        
Hit:5 http://de.archive.ubuntu.com/ubuntu focal InRelease                                                     
Hit:6 http://de.archive.ubuntu.com/ubuntu focal-updates InRelease                                             
Hit:7 http://de.archive.ubuntu.com/ubuntu focal-backports InRelease     
Hit:8 http://security.ubuntu.com/ubuntu focal-security InRelease        
Ign:9 http://dist.carla.org/carla focal InRelease                      
Err:10 http://dist.carla.org/carla focal Release
  404  Not Found [IP: 34.227.255.250 80]
Reading package lists... Done
E: The repository 'http://dist.carla.org/carla focal Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

qhaas, 2020-09-29:
It would appear that CARLA's [apt repo](http://dist.carla.org/carla/dists) currently only supports xenial (Ubuntu 16.04) and bionic (Ubuntu 18.04).  Have you tried using the [package](https://carla.readthedocs.io/en/latest/start_quickstart/#b-package-installation) (i.e. extract/run the tar.gz)?

shri-can, 2020-10-02:
> It would appear that CARLA's [apt repo](http://dist.carla.org/carla/dists) currently only supports xenial (Ubuntu 16.04) and bionic (Ubuntu 18.04). Have you tried using the [package](https://carla.readthedocs.io/en/latest/start_quickstart/#b-package-installation) (i.e. extract/run the tar.gz)?

Thanks, but never mind. I realised I needed to build from source so I just downgraded to Ubuntu 18.04. The 20.04 was a fresh install on my dual boot system anyway.

jhermen-List, 2020-10-21:
The 18.04 packages work on 20.04 without problems, just do a 
sudo add-apt-repository "deb [arch=amd64] http://dist.carla.org/carla bionic main"

togaen, 2020-10-30:
> The 18.04 packages work on 20.04 without problems, just do a
> sudo add-apt-repository "deb [arch=amd64] http://dist.carla.org/carla bionic main"

This worked for me on 20.04.

EsquilliDev, 2020-11-16:
So, the only "fix" is to use 18.04 packages?

togaen, 2020-11-25:
@EsquilliDev seems so at the moment. Unfortunately even though the sim runs, the python api doesn't work because the bundled egg file requires python 3.7, but 20.04 ships with 3.8.

EsquilliDev, 2020-11-25:
I was actually able to get it working by doing what was suggested [here](https://github.com/carla-simulator/ros-bridge/issues/415). I'm also using a conda environment with the right version of Python.

togaen, 2020-11-25:
Yeah, I can run a bridge (am using ROS 2), but do any of the util scripts under PythonAPI work for you? They do not work for me. They all give me import errors.

derFrickler, 2020-11-30:
It works fine for me on 20.04 if i just add the egg to the path manually before running the scripts:
export PYTHONPATH=$PYTHONPATH:/media/fast/LIST/PASCAL/PythonAPI/carla/dist/carla-0.9.10-py3.7-linux-x86_64.egg

togaen, 2020-11-30:
@derFrickler Do the util scripts work for you also?

derFrickler, 2020-11-30:
Yes, i am pretty sure i used them.
And wrote my own scripts based on the examples.
you might need to install the following packages as well:
apt-get install python3-numpy python3-pygame libxerces-c3.2

togaen, 2020-11-30:
I do have those packages. This is what I get when I try to run the `config.py` script:

```
$ ./config.py --help
Traceback (most recent call last):
  File "./config.py", line 28, in <module>
    import carla
  File "/opt/carla-simulator/PythonAPI/carla/dist/carla-0.9.10-py3.7-linux-x86_64.egg/carla/__init__.py", line 8, in <module>
  File "/opt/carla-simulator/PythonAPI/carla/dist/carla-0.9.10-py3.7-linux-x86_64.egg/carla/libcarla.py", line 9, in <module>
  File "/opt/carla-simulator/PythonAPI/carla/dist/carla-0.9.10-py3.7-linux-x86_64.egg/carla/libcarla.py", line 3, in __bootstrap__
ImportError: No module named util
```

togaen, 2020-11-30:
I realized the problem was the shebang in the util scripts... if I change them to invoke the correct python version, everything is fine. In `config.py`, for example, I just change the top of the file from:
```
#!/usr/bin/env python
```
to
```
#!/usr/bin/env python3
```

ll7, 2020-12-16:
How much effort is it to release a proper Ubuntu 20.04 package? As this is the first step in the [Quick start installation](https://carla.readthedocs.io/en/latest/start_quickstart/), some new users might be irritated?

qhaas, 2020-12-20:
> How much effort is it to release a proper Ubuntu 20.04 package?

In their defense, distro upgrades in enterprise environments isn't pleasant... heck, at my organization, we are still mostly using RHEL7 (contemporary with Ubuntu 14.04) on our production servers to host our headless CARLA runs...

jackz314, 2020-12-22:
> > How much effort is it to release a proper Ubuntu 20.04 package?
> 
> In their defense, distro upgrades in enterprise environments isn't pleasant... heck, at my organization, we are still mostly using RHEL7 (contemporary with Ubuntu 14.04) on our production servers to host our headless CARLA runs...

Having support for a new release doesn't mean they can't keep supporting the old versions. It shouldn't require too much effort since most of the problems only require minor version changes.

germanros1987, 2021-01-26:
@joel-mb could you please comment on this?

ll7, 2021-01-30:
> > How much effort is it to release a proper Ubuntu 20.04 package?
> 
> In their defense, distro upgrades in enterprise environments isn't pleasant... heck, at my organization, we are still mostly using RHEL7 (contemporary with Ubuntu 14.04) on our production servers to host our headless CARLA runs...

My question wasn't supposed to be criticism. I never released a package and I can't estimate the required effort. I am very grateful to the whole Carla team.

joel-mb, 2021-02-24:
Hi all,
I've added the latest CARLA release (0.9.11) in the debian repository for Ubuntu 20.04 (focal).

If you already setup the CARLA debian repository in your system it should work by running:
```bash
sudo apt-get update
sudo apt-get install carla-simulator
```

If you haven't setup the CARLA apt repository just follow the complete process:
```
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 1AF1527DE64CB8D9
sudo add-apt-repository "deb [arch=amd64] http://dist.carla.org/carla $(lsb_release -sc) main"
sudo apt-get update
sudo apt-get install carla-simulator
```

Please, take into account that, although everything should work well in Ubuntu 20.04, CARLA is only officially supporting Ubuntu 18.04.

closed 2021-03-03 by joel-mb

amandazhuyilan, 2021-06-25:
Note that for users with active firewalls, when setting up the gpg keys in the first step. you might want to do:
```
sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 1AF1527DE64CB8D9
```
instead.

CMakey, 2024-01-02:
sorry to distburn u, but is dist.carla.org still working now? 
several issues are called for this problems...
`(carla) lyc@lht-PowerEdge-R730:~/CARLA_0.9.9.4/PythonAPI/examples$ sudo add-apt-repository "deb [arch=amd64] http://dist.carla.org/carla focal main"
获取:1 file:/var/cudnn-local-repo-ubuntu2004-8.8.1.3  InRelease [1,572 B]
获取:1 file:/var/cudnn-local-repo-ubuntu2004-8.8.1.3  InRelease [1,572 B]
命中:2 http://mirrors.aliyun.com/ubuntu focal InRelease                                                
命中:3 http://mirrors.aliyun.com/ubuntu focal-security InRelease                                       
命中:4 http://mirrors.aliyun.com/ubuntu focal-updates InRelease                                        
命中:5 https://mirrors.tuna.tsinghua.edu.cn/ubuntu focal InRelease                                     
命中:6 https://mirrors.tuna.tsinghua.edu.cn/ubuntu focal-updates InRelease                             
命中:7 https://mirrors.tuna.tsinghua.edu.cn/ubuntu focal-backports InRelease                           
命中:8 http://mirrors.aliyun.com/ubuntu focal-proposed InRelease                                       
命中:9 http://mirrors.aliyun.com/ubuntu focal-backports InRelease                             
命中:10 http://security.ubuntu.com/ubuntu focal-security InRelease       
命中:11 http://ppa.launchpad.net/graphics-drivers/ppa/ubuntu focal InRelease
错误:12 http://dist.carla.org/carla bionic InRelease                                                   
  无法连接上 dist.carla.org:80 (34.227.255.250)，连接超时
错误:13 http://dist.carla.org/carla focal InRelease
  不能连接到 dist.carla.org：http：
正在读取软件包列表... 完成
W: 无法下载 http://dist.carla.org/carla/dists/bionic/InRelease  无法连接上 dist.carla.org:80 (34.227.255.250)，连接超时
W: 无法下载 http://dist.carla.org/carla/dists/focal/InRelease  不能连接到 dist.carla.org：http：
W: 部分索引文件下载失败。如果忽略它们，那将转而使用旧的索引文件。`
hope for u reply...

> 大家好，我已经在 Ubuntu 20.04 （focal） 的 debian 存储库中添加了最新的 CARLA 版本 （0.9.11）。
> 
> 如果您已经在系统中设置了 CARLA debian 存储库，它应该通过运行以下命令来工作：
> 
> ```shell
> sudo apt-get update
> sudo apt-get install carla-simulator
> ```
> 
> 如果您尚未设置 CARLA apt 存储库，请按照完整的过程操作：
> 
> ```
> sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 1AF1527DE64CB8D9
> sudo add-apt-repository "deb [arch=amd64] http://dist.carla.org/carla $(lsb_release -sc) main"
> sudo apt-get update
> sudo apt-get install carla-simulator
> ```
> 
> 请注意，尽管在 Ubuntu 20.04 中一切都应该运行良好，但 CARLA 仅正式支持 Ubuntu 18.04。

CMakey, 2024-01-02:
#7017