carla-simulator/carla #3376: [Testing] Dark sky

Madecu, 2020-09-29:
If you set the cloud amount above 50 and you lower the sun below 30º altitude, the sky turns pitch black. And at night, we had weird artifacts with the clouds turning greenish.

This has been solved for the 0.9.10 release. Can you check it out and play with the clouds and sun position to confirm it is no longer happening?

Thanks

XGodina, 2020-09-29:
I've tested on Windows and didn't see any artifact and nothing weird.

Madecu, 2020-09-29:
Tested in:

- Windows : Everything ok
- Linux Vulkan : Everything ok
- Linux OpenGL: The black sky and clouds issues are solved. But there is a problem with the sun reflection over puddles and water:

[image]

closed 2020-09-29 by Madecu