carla-simulator/carla #3377: Rendering Issues in 0.9.10

Kipt, 2020-09-29:
I decided to try out 0.9.10 and I noticed the colors seemed way too bright. After more investigation I also noticed severe glitches regarding reflections on water. By the looks of it the raycasting is failing horribly, causing black dots to appear.

The easiest way to see this is by setting weather to 'HardRainSunset' and looking at the reflections of the sun.

I'm running Carla through docker with using `-opengl` flag (without it I get a crash). I've tried using both **x11** and **offscreen** as `SDL_VIDEODRIVER` but no difference.

Full command is `docker run -e SDL_VIDEODRIVER=offscreen -it -p 2000-2002:2000-2002 --gpus all carlasim/carla:0.9.10 /bin/bash ./CarlaUE4.sh -opengl`.

The glitches are visible with spectator as well as through the sensors.

Tested with GeForce RTX 2080 Ti nvidia-driver-450
[image]

XGodina, 2020-09-29:
Hello @Kipt ,

We by default use Vulkan and the lighting values are set to this one. 
I don't know if that affects the docker. @bernatx can you help us with Docker?

p-testolina, 2020-10-01:
Hi guys,
I noticed the same brightness problem on 0.9.10 and 0.9.10.1 (tested this morning!). I'm running the docker image via singularity, and it's working fine without the `-opengl` flag, but the scenes seem way too bright. Same setup as @Kipt

XGodina, 2020-10-01:
Hi, @p-testolina,
Can you provide some images? Wich Town you tested?

p-testolina, 2020-10-01:
Hi @XGodina, thanks for the quick reply.
I tested the same weather settings using the `dynamic_weather.py` example using Town03. I hope I got them right.

This is what I get from the spectator with 0.9.9
[image]
and with 0.9.10
[image]
I didn't have the time to check the camera acquisitions yet. I can produce them if you need them.

Madecu, 2020-10-02:
Hi @p-testolina,
In the new 0.9.10, we have changed the exposure mode of the camera to auto exposure and we have tweaked the camera configuration a little bit. That is why the image is now brighter than before.
We have exposed a lot of parameters so you can customize your RGB camera the same way you would in real life. You can set your camera (the RGB sensor of the car or the Spectator camera from unreal) with your customize ISO, aperture, shutter speed, and more advanced configurations. Check out the documentation, we listed all the parameters for the RGB sensors:
[https://carla.readthedocs.io/en/latest/ref_sensors/#rgb-camera](https://carla.readthedocs.io/en/latest/ref_sensors/#rgb-camera)

Regarding the issue @Kipt  has with those black reflections, we are aware of that. It only happens on Linux with OpenGL. We are working on that. We can also try to solve why it crashes if you use Vulkan instead, so you don't have to use the opengl flag. OpenGL doesn't support other features such as mesh distance fields, so it would be better to use vulkan if you can.

Kipt, 2020-10-05:
It seems adding some parameters to docker fixed the issue. With `--gpus 'all,"capabilities=graphics,utility,display,video,compute"'` it runs with Vulkan resulting in the below image. Still quite brighter than the 0.9.9 spectator camera but as the brightness can be modified in the camera sensors it's a minor detail.

I suggest adding a note in the documentation https://carla.readthedocs.io/en/latest/build_docker/.

Thank you for your help, feel free to close if the above docker parameter looks correct.

[image]

XGodina, 2020-10-05:
Hi, @Kipt
Thank you for your input

closed 2020-12-25 by stale[bot]

qhaas, 2021-01-25:
Seeing this issue with opengl, bare-metal (i.e. no container) in Ubuntu 20.04 with opengl on a dgx a100 under certain weather conditions (e.g. HardRainSunset).
[image]