carla-simulator/carla #3385: Some sensors ignore sensor_tick attribute [0.9.10 / 0.9.10.1]

jorothe, 2020-10-01:
Hi,

the sensors _camera.rgb, camera.semantic_segmentation_ and _camera.depth_ seem to ignore the sensor_tick attribute and return data in every tick.

The _sensors camera.dvs_ or _lidar.ray_cast_ for example are behaving as expected.

I am using Carla 0.9.10[.1] on Windows.

guntherkrehl, 2020-10-06:
Hi there,

I am experiencing the exact same issue with the SceneCaptureSensors. I've already contacted @marcgpuig and he told me the team (or he himself) is on it. :) Just wanted to bump this thread with this comment.

Best,
Gunther

marcgpuig, 2020-10-13:
Hi @jorothe @guntherkrehl! Sorry for the late response! 
This should be fixed in `dev` branch since #3399, please, if you still can reproduce it, make sure to report it here #3400.
Thanks :)

closed 2020-10-30 by marcgpuig