carla-simulator/carla #3391: Inconsistent frame/elapsed time for camera sensor data in 0.9.10-2

analog-cbarber, 2020-10-02:
I am seeing inconsistent frame numbers and elapsed time values using the latest 0.9.10-2 debian release on Ubuntu.

I have scenario where I am placing random pedestrians in each frame, and viewing them with three co-located cameras: rgb, depth and segmentation. I am using synchronous mode with a fixed time delta of .1s (10fps). This worked ok with 0.9.10-1,
but with the latest release I am sometimes getting consecutive records from each sensor that have the same frame number or same elapsed time as the previous frame and if you look at the images, they don't line up consistently. So I basically cannot trust the frame and timestamp labels I am getting from Carla.

Here are a few frames from the segmentation and rgb cameras. In this case, the segmentation camera frame numbers are one less than the corresponding rgb frames and the elapsed times are either one behind or are just wrong. But there is no consistent pattern even within a single episode.

[image]

I don't have a simple script to reproduce this right now, but I can write one if you need it.

Is this a known issue?

analog-cbarber, 2020-10-05:
Here is a script to reproduce at least some of the bad behavior.

Using this script, I have not observed a total mismatch of images with the same frame #'s as was shown in the above example, but it does usually show duplicate frames and mismatches with frames with the same elapsed timestamp. I suspect the reason my own code is showing worse effect is due to some difference in timing.

Unless there is some workaround for this, I don't think I can do any sensor fusion with the data from this version of Carla.

```python
"""
Script to reproduce Carla bug 3391
"""

import argparse
import csv
import os
import re
import subprocess
import sys
import time
from abc import ABC, abstractmethod
from glob import glob
from queue import Queue, Empty
from shutil import rmtree
from threading import Thread
from typing import cast

carla_root = os.getenv('CARLA_PYTHON_ROOT', '/opt/carla-simulator')
assert os.path.isdir(carla_root)
glob_expr = os.path.join(carla_root, 'PythonAPI', 'carla', 'dist', 'carla-*-py3.7-*.egg')
print(glob_expr)
carla_egg = glob(glob_expr)
sys.path.append(carla_egg[0])

import carla
import numpy as np

class CarlaSession:
    server_process = None
    client = None

    def __init__(self):
        pass

    def start(self):
        self.server_process = subprocess.Popen(
            [os.path.join(carla_root, 'CarlaUE4','Binaries', 'Linux', 'CarlaUE4-Linux-Shipping'), 'CarlaUE4'],
            stdout=subprocess.DEVNULL, stderr=subprocess.DEVNULL
        )
        time.sleep(5)
        self.client = carla.Client('localhost', 2000)

    def stop(self):
        proc = self.server_process
        proc.terminate()
        for i in range(20):
            try:
                proc.wait(1)
            except (TimeoutError, subprocess.TimeoutExpired):
                print(i)
                proc.kill()

    def __enter__(self):
        self.start()
        return self

    def __exit__(self, exc_type, exc_val, exc_tb):
        self.stop()

class Camera(ABC):
    name:str
    input_queue:Queue
[... 246 lines cut ...]
                continue # no pedestrians in view

            # find corresponding depth frame
            for row in depth_frames:
                depth_frame = int(row[1])
                depth_elapsed = float(row[2])
                if depth_elapsed == elapsed:
                # if depth_frame == frame:
                    dn = int(row[0])
                    depthn = np.load(depthcam.data_file(dn))
                    depthn[pedmask] = depth0[pedmask]
                    # the pixels without pedestrians in them should have the same values
                    # really they should be the same, but we will allow a fudge factor
                    norm = np.linalg.norm(depthn - depth0)
                    isclose = norm < 10
                    # if isclose:
                    #     print(f'Depth and seg at frame {frame} are ok')
                    if not isclose:
                        print(f'Depth {dn} and seg {n} at frame {frame} {elapsed:.2f}s are mismatched',
                              file=sys.stderr)
                    break

if __name__ == '__main__':
    main()
```

thytless, 2020-10-12:
Hello! Did your problem solved in the latest release 0.9.10.1?

PTorr, 2020-11-23:
I'm not sure, but I think your problem is that you use wold.tick() multiple times there.

It should be use once for each iteration of the synchronous mode.

closed 2021-01-26 by germanros1987