carla-simulator/carla #3392: Framerate cap and no-rensering mode

pomputer, 2020-10-03:
Hi!
Is it possible to configure a framerate cap (60fps) in order to not waste resources for rendering useless frames. Also, is it possible to have the server in no-rendering mode, but the clients in normal rendering mode?

Thanks!

Derekabc, 2020-11-07:
Hi Promputer, any progress on this problem?

pomputer, 2020-11-08:
Sadly, no.

pomputer, 2020-11-08:
> Hi Promputer, any progress on this problem?

Sadly, no.

germanros1987, 2021-01-26:
@pomputer, you can set the simulation to synchronous mode, and then you can tick from the client-side at your preferred speed to produce the cap.

germanros1987, 2021-01-26:
+@DSantosO to provide help

DSantosO, 2021-02-01:
Hello @pomputer, @Derekabc,
As @germanros1987 explained, you can set the simulator in synchronous mode and set the delta seconds (1/fps) that you want. You can find all the information [here](https://carla.readthedocs.io/en/latest/adv_synchrony_timestep/). 
In the same world settings, you can check the `no_rendering_mode = True` to do what you need.

closed 2021-02-02 by pomputer