carla-simulator/carla #3398: Carla 0.9.9.10 Crashes after hitting Play

leanderpeter, 2020-10-06:
Hi guys,

first of all i love the new possibilities with Carla 0.9.9.10. I’m very thrilled to test out the many features.

**The problem:**

After building Carla from source (strictly followed the Linux docs 3 times) Carla crashes after hitting the “Play Button” in the Unreal Editor. Process: -> make Python API (works) -> make launch (works) -> press play after all meshes and shaders are ready (loud CPU noises, then crash). I get the following Error Message:

```
[2020.10.05-17.02.43:959][178]LogSkeletalMesh: Warning: Skeletal ComputeTangents MikkTSpace function: Compute a zero length normal vector.
/home/l/carla/Util/BuildTools/BuildCarlaUE4.sh: line 149: 10302 Killed                  ${GDB} ${UE4_ROOT}/Engine/Binaries/Linux/UE4Editor "${PWD}/CarlaUE4.uproject" ${RHI}
Util/BuildTools/Linux.mk:7: recipe for target 'launch' failed
make: *** [launch] Error 137

```
**My setup:**

-     Ubuntu 18.04
-     AMD 3600X
-     NVIDIA rtx 2070 (NVIDIA driver metapackage 435 proprietary)
-     16GB RAM

leanderpeter, 2020-10-14:
Problem solved: swap partition to small (1GB). Now 17 GB, working fine.

closed 2020-10-14 by leanderpeter

goksanisil23, 2020-11-28:
Hi @leanderpeter , thanks for indicating your solution, I had the exact same issue and increasing the swap space to 8GB worked. 
May I ask, why is the swap space important in this case?
When I run "top", there are parallel shaders being compiled when you press play in Unreal. My CPU load increases to 800% but as far as I know increasing having a swap partition helps in the cases where the RAM is not sufficient. In this case, is it use to transfer data to GPU's memory?

I have NVIDIA 1660Ti,   Ryzen5 2600x, 16GB RAM

Thanks and regards

leanderpeter, 2020-11-28:
I can only assume. The CPU processes things in memory. So if you are processing a lot of data, this data will probably be loaded into RAM so that it is ready for the CPU to speed things up, because RAM can be accessed faster than memory devices. My assumtion is that processing the shaders, RAM usage peaks and Linux will start to swap.

goksanisil23, 2020-11-29:
Thanks for sharing your intuition, otherwise would've took me long time to realize 16GB RAM is not sufficient

lydiaCS, 2021-03-15:
RAM usage might up to 5GB and increasing the swap space to 8GB workes for me. thanks!