carla-simulator/carla #3402: Make Launch failure for latest build 0.9.10

SSubhnil, 2020-10-06:
Version: 0.9.10
OS: Win 10 (latest)
GPU: GTX 1050Ti 4 GB

I have the latest assets and `dev` branch code.
I had errors with Unreal Editor after building from 0.9.9 to 0.9.10 so I am now building 0.9.10 in a different drive. The `make PythonAPI` encounters following message:

```
CMake Error at D:/Program Files/CMake/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:165 (message):
  Failed to find XercesC (missing: XercesC_VERSION)
Call Stack (most recent call first):
  D:/Program Files/CMake/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:458 (_FPHSA_FAILURE_MESSAGE)
  D:/Program Files/CMake/share/cmake-3.18/Modules/FindXercesC.cmake:103 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:86 (find_package)

-- Configuring incomplete, errors occurred!
See also "E:/Carla/carla-dev/Build/osm2odr-visualstudio/CMakeFiles/CMakeOutput.log".
Microsoft (R) Build Engine version 15.9.21+g9802d43bc3 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

MSBUILD : error MSB1009: Project file does not exist.
Switch: install.vcxproj
-[BuildOSM2ODR]: OSM2ODR has been successfully installed in "E:\Carla\carla-dev\PythonAPI\carla\dependencies\"
-[BuildPythonAPI]: [Batch params]: --py3
Building Python API for Python 3.
Traceback (most recent call last):
  File "setup.py", line 163, in <module>
    ext_modules=get_libcarla_extensions(),
  File "setup.py", line 119, in get_libcarla_extensions
    libs = [x for x in os.listdir('dependencies/lib') if any(d in x for d in required_libs)]
FileNotFoundError: [WinError 3] The system cannot find the path specified: 'dependencies/lib'

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "E:\Carla\carla-dev\PythonAPI\carla\dist"!
```
It never creates a `dist` folder in `PythonAPI\carla` but is somehow successful.

Then when I do `make launch`, I get the following:
```
 [12/21] Module.CarlaExporter.cpp
  E:/Carla/carla-dev/Unreal/CarlaUE4/Plugins/CarlaExporter/Source/CarlaExporter/Private/CarlaExporter.cpp(153): warning C4996: 'AActor::GetComponentsByClass': Use one of the GetComponents implementations as appropriate Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  E:\Program Files\UE_4.24\Engine\Source\Runtime\Engine\Classes\GameFramework/Actor.h(2881): note: see declaration of 'AActor::GetComponentsByClass'
  [13/21] UE4Editor-CarlaExporter.lib
     Creating library E:\Carla\carla-dev\Unreal\CarlaUE4\Plugins\CarlaExporter\Intermediate\Build\Win64\UE4Editor\Development\CarlaExporter\UE4Editor-CarlaExporter.lib and object E:\Carla\carla-dev\Unreal\CarlaUE4\Plugins\CarlaExporter\Intermediate\Build\Win64\UE4Editor\Development\CarlaExporter\UE4Editor-CarlaExporter.exp
  [14/21] Module.Carla.cpp
  E:/Carla/carla-dev/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Commandlet/PrepareAssetsForCookingCommandlet.cpp(58): warning C4458: declaration of 'TerrainNodeMaterial' hides class member
  e:\carla\carla-dev\unreal\carlaue4\plugins\carla\source\carla\commandlet\PrepareAssetsForCookingCommandlet.h(161): note: see declaration of 'UPrepareAssetsForCookingCommandlet::TerrainNodeMaterial'
  E:/Carla/carla-dev/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(505): warning C4458: declaration of 'Role' hides class member
  E:\Program Files\UE_4.24\Engine\Source\Runtime\Engine\Classes\GameFramework/Actor.h(534): note: see declaration of 'AActor::Role'
  E:/Carla/carla-dev/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(514): warning C4458: declaration of 'Role' hides class member
  E:\Program Files\UE_4.24\Engine\Source\Runtime\Engine\Classes\GameFramework/Actor.h(534): note: see declaration of 'AActor::Role'
  E:/Carla/carla-dev/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorderQuery.cpp(46): warning C4996: 'localtime': This function or variable may be unsafe. Consider using localtime_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.
  C:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt\time.h(506): note: see declaration of 'localtime'
  E:/Carla/carla-dev/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/CollisionSensor.cpp(49): warning C4458: declaration of 'Episode' hides class member
  E:\Carla\carla-dev\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Sensor/Sensor.h(99): note: see declaration of 'ASensor::Episode'
  E:/Carla/carla-dev/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/DVSCamera.cpp(143): warning C4458: declaration of 'Stream' hides class member
  E:\Carla\carla-dev\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Sensor/Sensor.h(97): note: see declaration of 'ASensor::Stream'
  E:/Carla/carla-dev/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/GnssSensor.cpp(51): warning C4458: declaration of 'Stream' hides class member
  E:\Carla\carla-dev\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Sensor/Sensor.h(97): note: see declaration of 'ASensor::Stream'
  E:/Carla/carla-dev/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/InertialMeasurementUnit.cpp(48): warning C4458: declaration of 'Owner' hides class member
  E:\Program Files\UE_4.24\Engine\Source\Runtime\Engine\Classes\GameFramework/Actor.h(516): note: see declaration of 'AActor::Owner'
  E:/Carla/carla-dev/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/InertialMeasurementUnit.cpp(184): warning C4458: declaration of 'Stream' hides class member
  E:\Carla\carla-dev\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Sensor/Sensor.h(97): note: see declaration of 'ASensor::Stream'
  E:/Carla/carla-dev/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/ObstacleDetectionSensor.cpp(138): warning C4458: declaration of 'Episode' hides class member
  E:\Carla\carla-dev\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Sensor/Sensor.h(99): note: see declaration of 'ASensor::Episode'
  E:/Carla/carla-dev/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Server/CarlaServer.cpp(19): fatal error C1083: Cannot open include file: 'carla/Version.h': No such file or directory
  [15/21] UE4Editor-CarlaExporter.dll
     Creating library E:\Carla\carla-dev\Unreal\CarlaUE4\Plugins\CarlaExporter\Intermediate\Build\Win64\UE4Editor\Development\CarlaExporter\UE4Editor-CarlaExporter.suppressed.lib and object E:\Carla\carla-dev\Unreal\CarlaUE4\Plugins\CarlaExporter\Intermediate\Build\Win64\UE4Editor\Development\CarlaExporter\UE4Editor-CarlaExporter.suppressed.exp
  [16/21] CarlaUE4.cpp
  [17/21] UE4Editor-CarlaUE4.lib
     Creating library E:\Carla\carla-dev\Unreal\CarlaUE4\Intermediate\Build\Win64\UE4Editor\Development\CarlaUE4\UE4Editor-CarlaUE4.lib and object E:\Carla\carla-dev\Unreal\CarlaUE4\Intermediate\Build\Win64\UE4Editor\Development\CarlaUE4\UE4Editor-CarlaUE4.exp
make: *** [CarlaUE4Editor] Error 5
```
I cannot really figure what error 5 means.

nagasanjay, 2020-10-09:
please cross-verify the python version you are currenty using ( type python in cmd and look at the version ) and the version mentioned in the file named `libboost` in the folder `PythonAPI\carla\dependencies\lib\` .

If they vary, install the version mentioned in that `libboost` file and then make it the default python version ( keep it at the top of environment variables' `path` variable list ).

1. try `make launch` again. If it didn't work,
2. try running `python setup.py install` in the folder `PythonApi\carla`.

charlie-wartnaby, 2020-11-24:
I have the same problem building 0.9.10.1 under Windows. Previously I had a similar problem with 0.9.9 under Linux (see #3163). The problem is libxerces build fails, because cmake doesn't find it when invoked by BuildOSM2ODR.bat (as shown in your error output):`

```Failed to find XercesC (missing: XercesC_VERSION)```

 I'm trying to figure out how to get past this; I've found some tips online about giving hint paths to cmake. I wonder if this commit (last one of several recently) might be relevant?

```commit c2cb39afca552b5533f31fd35c8fce4cf655f613
Author: Axel <axellopez92@outlook.com>
Date:   Wed Sep 30 09:42:52 2020 +0200

    Removed libxerces-c from prerequisites and documentation
```
Charlie Wartnaby, Applus IDIADA

charlie-wartnaby, 2020-11-24:
OK this was a bit of a Heisenbug to me :-) .

I first peppered `install_xercesc.bat` with ECHO and PAUSE statements so that I could step through it and see what was going on.
I also deleted `carla/Build/xerces*` (I had 3 folders) completely, to force a rebuild of xerces-c.
When I ran 'make PythonAPI' again, the xerces-c build succeeded this time, generating `carla\Build\xerces-c-3.2.3-install\lib\xerces-c_3.lib`.
Then I deleted my reverted out my changes to `install_xercesc.bat`, deleted `carla/Build/xerces*`, and again it worked.

However, I have a theory. I have had problems with the other `install_*.bat` files, which I'm guessing is due to the action of Kaspersky Endpoint Security on my corporate machine. Those .bat files tend to fail with Access Denied errors when they attempt to delete or rename a folder which they have just created, e.g. by expanding a .zip file. I'm guessing that's because Kaspersky is still scanning the newly-created files, and it has some kind of lock on them (which it probably shouldn't). I fixed those by adding TIMEOUT commands to the .bat files before any folder renames or deletes, for several minutes in the case of the enormous boost installation.

So I suggest you try this:
1. Just delete `carla/Build/xerces*` and try `make PythonAPI` again. You might get lucky this time. 
2. If you're not lucky, look carefully for errors generated by `install_xercesc.bat`
3. You could try this modification which I've just put in my version of that file, which (this time) wasn't necessary, but fits the pattern of the other .bat files I had to modify to work on this PC:

```
    ...
    rem Remove unnecessary files and folders
    echo %FILE_N% Removing "%XERCESC_TEMP_FILE%"

    echo CW IDIADA pausing a while in case Kaspersky is still scanning files we are about to delete/rename...
    timeout 60

    del "%XERCESC_TEMP_FILE_DIR%"
    ...
```

As for that recent change to remove libxerces-c from the prerequisites, I understand now that it has been added as one of the packages built from source, hence no need for it to be separately installed any more, so that fits.

All working for me now (except a mysteriously blank pygame window when I run `manual_control.py` as a test -- not related, I'm sure).

Charlie Wartnaby, Applus IDIADA

germanros1987, 2021-01-26:
@bernatx could you provide some feedback?

huatianxue1994, 2021-05-26:
> please cross-verify the python version you are currenty using ( type python in cmd and look at the version ) and the version mentioned in the file named `libboost` in the folder [...]

I am sorry that there is not a folder `PythonAPI\carla\dependencies\lib\`,I do not know why. So I can not find a  file named "libboost". There is only such file as follow:
[image]
The error like follow:
[image]

huatianxue1994, 2021-05-26:
> please cross-verify the python version you are currenty using ( type python in cmd and look at the version ) and the version mentioned in the file named `libboost` in the folder [...]

And When I make launch, the error is like follow:
[image]
What should I do? Thank you advanced!

junaidbinsarfraz, 2022-11-25:
Still getting the same error
Windows 10 (latest)
Carla: 0.9.13
Command: `make PythonAPI`

Error:
[image]

Also, [this comment](https://github.com/carla-simulator/carla/issues/3402#issuecomment-706346916) suggests checking the Python version from `libboost` file in the folder `PythonAPI\carla\dependencies\lib\`, but this file doesn't exist in this folder.

winterwindwang, 2023-02-10:
> Still getting the same error Windows 10 (latest) Carla: 0.9.13 Command: `make PythonAPI`
>
> Error: [image]
>
> Also, [this [...]

Did you solve this problem?

EmmettJZK, 2023-02-17:
> Did you solve this problem?

Hi,guys. I met the same problem recently. But today I knew why it happened in my environment. 
My Environment: 
    Windows10 Pro
    Python3.7.9
    Carla 0.9.14
    UE-Carla master branch
At beginning, I downloaded zip file of carla 0.9.14 due to low speed with git. I'm not confirmed that will influnce the compiling process. But when I met the problem, what I did is redownloading the project with git command and recompiling the project again with command "make PythonAPI". And this time, the error was more understandable: "error: invaild command 'bdist_wheel'" which was solved in issue #5542. Seeing the solution mentioned in #5542, I realized I forgot to install 'wheel' python package in former step. So of course, I cannot build wheel file in PythonAPI/carla/dist folder. But I dont know why this time, I successed building the libs in the folder \dependencies\lib. 
Hope that can help you all.

closed 2024-05-10 by MarcelPiNacy-CVC