carla-simulator/carla #3403: Can I launch Carla on a server via ssh?

AminoACID123, 2020-10-07:
I have a server in my school lab that has CARLA installed, and I want to use my laptop to remotely manipulate it.

So I used ssh to connect to the server, but when I ran `./CarlaUE4.sh -vulkan` command, nothing happened. CARLA on the server did not start. Nor did I see anything on my laptop screen.

So could someone tell me how to launch CARLA remotely? thx.

HemaZ, 2020-10-07:
You can run CARLA in an Off-screen mode using the command 
`DISPLAY= ./CarlaUE4.sh -opengl
`
https://carla.readthedocs.io/en/latest/adv_rendering_options/#off-screen-mode

closed 2020-10-09 by AminoACID123

ChristofKaufmann, 2021-09-22:
For anyone finding this. The info above is outdated. In version 0.9.12 use this (see [here](https://carla.readthedocs.io/en/latest/adv_rendering_options/#off-screen-mode)):
```bash
./CarlaUE4.sh -RenderOffScreen
```