carla-simulator/carla #3410: Question about switching controller (eg. switch from keyboard to steering wheel)

shiyuanzhu123, 2020-10-08:
Hi, I'm running the "manual_control.py" script (in example folder) in my own map and  I want to achieve one function: switch driving method from keyboard to steering wheel by pressing one button. Could you please give me some instructions on that? Thanks a lot!

Yuan Shi

seuwcs, 2020-10-09:
if you had connected a steering wheel to carla?

seuwcs, 2020-10-09:
But I am wondering how did you solve the problem, as I met the same problem.  Would you mind giving me some suggestions regarding this question. Very much thank!
I had a steering wheel but it can't be connected to Carla in Win10, How it can be solved?

jadesop, 2020-12-07:
Hello, I'm curious about your steering wheel, do you use the Logitech steering wheel? Or series of Thrustmaster?

closed 2021-01-26 by germanros1987