carla-simulator/carla #3412: Example High FOV Lens Distortion Parameters

jimmyw404, 2020-10-08:
Thank you for making a Lens Distortion model! However I am having trouble using it to generate the kind of lens distortions I typically see in real-world lenses with high fields of view.

I am trying to simulate a wide angle lens and a fisheye lens. I understand that 180 degrees are annoying to render, so I don't need to go that high. 160 degrees is enough.

Below is a sample of a wide angle 160 degree lens from https://www.nytimes.com/wirecutter/reviews/best-dash-cam/

[image]

And here is a sample of a simulated fisheye lens (shown inset) from https://medium.com/cruise/cruise-simulation-sensor-development-be57a5991fe6

[image]

I tried quite a few different options to get this kind of lens distortion but everything I tried distorted it in unrealistic ways. I used this documentation to try and figure it out:

https://carla.readthedocs.io/en/latest/ref_sensors/#rgb-camera

[image]

Below is the distorted camera in the manual_control.py with a fov of 160:

[image]

Would it be possible to get example parameters for the lens distortion that would produce something similar to the above wide angle / fisheye lens images? Or should I pursue a different lens model (ex: https://docs.unrealengine.com/en-US/API/Plugins/LensDistortion/FLensDistortionCameraModel/index.html or opencv lens distortion plugin: https://docs.unrealengine.com/en-US/API/Plugins/OpenCVLensDistortion/index.html )
        

@marcgpuig

jimmyw404, 2020-10-08:
#776 Lens distortion shader
#2145 Lens distortion shader
#754 Lens distortion shader
#2669 Fisheye camera support in carla
#3130 Lens distortion attributes
#2048 Lens Distortion

jimmyw404, 2020-11-13:
I'm still very interested in guidance on this issue!

germanros1987, 2021-01-26:
@DSantosO @doterop I would like to provide an answer to this issue. Please, work on this.

YifuTao, 2021-02-17:
Hi, I would be very interested as well in the fisheye camera model / hige fov camera by the distortion model. 

I found the [distortion model documentation ](https://carla.readthedocs.io/en/latest/ref_sensors/#rgb-camera) does not currently provide enough information on guiding that. This is also mentioned also [in this issue](https://github.com/carla-simulator/carla/issues/3130#issuecomment-780353448)

closed 2021-07-21 by stale[bot]

Glazyee, 2021-09-29:
> Hi, I would be very interested as well in the fisheye camera model / hige fov camera by the distortion model.
>
> I found the [distortion model documentation [...]

Hi, FYI, the RGB camera seems based only on the Scene Capture Component 2D, which doesn't include a distortion model. You might need to write a post-processing script to introduce the distortion or either create a blueprint using any other cinematic camera blueprint available in unreal engine editor.

mittkongg, 2022-02-18:
> Thank you for making a Lens Distortion model! However I am having trouble using it to generate the kind of lens distortions I typically see in real-world lenses with high fields of view.
>
> I am [...]

Does k1 k2 k3 have anything to do with lens_circle_falloff lens_circle_multiplier？

jimmyw404, 2022-02-18:
> Does k1 k2 k3 have anything to do with lens_circle_falloff lens_circle_multiplier？

I don't know. I don't know of any way to relate the lens distortion in CARLA to anything like OpenCV or real camera lenses.

yuanhaorannnnnn, 2022-07-28:
@jimmyw404 Hi, I'm facing the same problem and just find this page, while at first I've a naive question. When you changed the distoration parmameters of the sensor.camera.rgb, and you can see the difference with running the manual_control.py, how did you make that? Which file you change to do this?
Many thanks in advance.

danieleceUL, 2022-09-23:
In my case, CARLA doesn't have a distorted RGB view when I press number 8 to switch to camera distortion. Is it possible to find the example script for what is happening here?

jimmyw404, 2022-09-23:
@yuanhaorannnnnn  @danieleceUL  

in this file: https://github.com/carla-simulator/carla/blob/master/PythonAPI/examples/manual_control.py#L1066

```
            ['sensor.camera.rgb', cc.Raw, 'Camera RGB Distorted',
                {'lens_circle_multiplier': '3.0',
                'lens_circle_falloff': '3.0',
                'chromatic_aberration_intensity': '0.5',
                'chromatic_aberration_offset': '0'}],
```

Change the fov to some high value, ex: 160 as per https://carla.readthedocs.io/en/latest/ref_sensors/#rgb-camera ex:

```
            ['sensor.camera.rgb', cc.Raw, 'Camera RGB Distorted',
                {'lens_circle_multiplier': '3.0',
                'lens_circle_falloff': '3.0',
                'chromatic_aberration_intensity': '0.5',
                'chromatic_aberration_offset': '0',
                'fov':'160'}],
```

danieleceUL, 2022-09-23:
Thank you @yuanhaorannnnnn. I tried it out there looks great!

danieleceUL, 2022-09-23:
Sorry it was you @jimmyw404 who was so helpful! :-)

abhay-707, 2025-05-07:
> Hello, Please, have you found an answer for this question? what is the relationship between OpenCV distortion model and CARLA model?

I would like to know the answer of this question