carla-simulator/carla #3413: Regarding the installation of Carla with Pip

kartikeyshaurya, 2020-10-09:
getting this error while installing the carla with pip :- anybody have idea why it is occurring 

 ERROR: Command errored out with exit status 1 , 
``     
command: 'C:\Users\karti\miniconda3\python.exe' -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\karti\\AppData\\Local\\Temp\\pip-install-kqh8f8st\\carla\\setup.py'"'"'; __file__='"'"'C:\\Users\\karti\\AppData\\Local\\Temp\\pip-install-kqh8f8st\\carla\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base 'C:\Users\karti\AppData\Local\Temp\pip-install-kqh8f8st\carla\pip-egg-info'
         cwd: C:\Users\karti\AppData\Local\Temp\pip-install-kqh8f8st\carla\
    Complete output (7 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\karti\AppData\Local\Temp\pip-install-kqh8f8st\carla\setup.py", line 114, in <module>
        ext_modules=get_libcarla_extensions(),
      File "C:\Users\karti\AppData\Local\Temp\pip-install-kqh8f8st\carla\setup.py", line 75, in get_libcarla_extensions
        libs = [x for x in os.listdir('dependencies/lib') if any(d in x for d in required_libs)]
    FileNotFoundError: [WinError 3] The system cannot find the path specified: 'dependencies/lib'
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
``

qhaas, 2020-10-13:
Try `easy_install` instead

kartikeyshaurya, 2020-10-13:
> Try `easy_install` instead. I have built CARLA from source using `bdist_wheel` and tried to deploy the resulting .whl file only to find it doesn't find the CARLA library, yet the eggs built from source work fine when deployed with `easy_install`. However, when using the `pip3` from the Ubuntu 18.04 repository (i.e. NOT conda), the whl/pip3 work fine...

It's Working reinstalled the conda version,but the Above one also be Working because the problem was in the bdist Wheel

closed 2020-10-20 by kartikeyshaurya