carla-simulator/carla #3415: Error when convert OSM to XODR

siraphopp, 2020-10-09:
## My code:

f = open('map.osm', 'r')
osm_data = f.read()
f.close()

settings = carla.Osm2OdrSettings()
xodr_data = carla.Osm2Odr.convert(osm_data, settings)

## Error:

RuntimeError: unable to open file 'https://sumo.dlr.de/xsd/types_file.xsd'
 In file 'built in type map'
 At line/column 1/0.

  The types could not be loaded from 'built in type map'.

## My OSM map:

[map.osm.zip](https://github.com/carla-simulator/carla/files/5354125/map.osm.zip)

jakubLangr, 2020-10-16:
Hi @siraphopp we have the same issue—did you ever figure out the solution?

ssuralcmu, 2020-11-01:
I am trying to load an OSM file to generate a Carla Map but have the exact same error. Has anyone figured this out yet?

LVJ047, 2020-11-10:
I am also trying to obtain the same results but I am getting the following error while using Carla 0.9.10 :
[image]

karolmajek, 2021-01-22:
This was my error:
```
Warning: Cannot read local schema '../carla/data/xsd/types_file.xsd', will try website lookup.
Traceback (most recent call last):
  File "convert_osm2carla.py", line 17, in <module>
    xodr_data = carla.Osm2Odr.convert(osm_data, settings)
RuntimeError: unsupported protocol in URL
 In file 'built in type map'
 At line/column 1/0.

  The types could not be loaded from 'built in type map'.
```

# Solved it!!!

```
Warning: Cannot read local schema '../carla/data/xsd/types_file.xsd', will try website lookup.
```

run this from `CARLA_0.9.11/PythonAPI/examples` directory

closed 2021-01-26 by germanros1987

munirjojoverge, 2021-05-12:
I don't see how this has been solved? what did you run?