carla-simulator/carla #3417: The carla map is too white when opened in UE4 editor

nagasanjay, 2020-10-09:
I downloaded the carla and built it from source. I wanted to edit the map, so I opened it in UE editor. But it looks only white and I can't see it properly.

I use windows 10. Carla version 0.9.10. Unreal engine version 2.24.3.
[image]

Also, I couldn't connect to the server (after clicking play button in UE editor). I asks me to check whether the server is binded to port 2000.

Madecu, 2020-10-13:
Hi @nagasanjay, that is weird. There might be something outdated. If you go inside the BP_Sky in your scene, there is a PostprocessingVolume. Can you check if the exposure and camera options are the same that in this picture?

[image]

nagasanjay, 2020-10-14:
@Madecu Thanks for replying... No the values weren't the same and I figured out why.

For the first time when I ran the CarlaUE4 editor, it took me too long at **39%** (in welcome screen). I searched about it and in one similar issue, someone suggested me to delete the `config` folder, which I did.

That's why it wasn't working. I downloaded the config files again and moved them to config folder, which solved my issue...

closed 2020-10-14 by nagasanjay