carla-simulator/carla #3419: [Question] How to draw waypoints of the route a vehicle took in a recording

K-Hoppe, 2020-10-10:
Hello, I want to show the route my vehicle takes in a recording and need a way to draw just these waypoints. Is there a way to draw the waypoints of a vehicle?

nagasanjay, 2020-10-10:
refer to this [issue](https://github.com/carla-simulator/scenario_runner/issues/40).
It shows how you can save the location of the spectator. In a similar way, you can try to get the location of the actor.

K-Hoppe, 2020-10-10:
> 
> 
> refer to this [issue](https://github.com/carla-simulator/scenario_runner/issues/40).
> you can get the location of the vehicle by calling `vehicle.get_location()`

Thanks for your answer, but thats not what iI wanted to do.
I want the waypoints drawn where the vehicle was/will be. 
Or
I want to draw the route the vehicle took

nagasanjay, 2020-10-10:
Correct me if I'm wrong. Won't you get the waypoints if you call get_location() continuously, for each tick ?

mercurylaci, 2020-10-12:
This is an example, based on this I think you can do it.
begin = carla.Location(x=x1, y=y1, z=0.1)
world.debug.draw_point(begin, size=0.05, life_time=7200, color=carla.Color(238, 18, 137, 0))

K-Hoppe, 2020-10-12:
Thanks! I figured it out.

closed 2020-10-12 by K-Hoppe