carla-simulator/carla #3420: fatal error: too many errors emitted, stopping now for rpclib library

NishanthARao, 2020-10-10:
Hello all. Could you please help me figure out why I am getting this error?
Carla Version: 0.9.10
Ubuntu: 18.04 Bionic Beaver
Installation link that I am following: https://carla.readthedocs.io/en/latest/build_linux/
```bash
Setup.sh: Retrieving rpclib.
Cloning into 'rpclib-v2.2.1_c3-c8-source'...
remote: Enumerating objects: 14, done.
remote: Counting objects: 100% (14/14), done.
remote: Compressing objects: 100% (9/9), done.
remote: Total 7006 (delta 2), reused 14 (delta 2), pack-reused 6992
Receiving objects: 100% (7006/7006), 3.29 MiB | 620.00 KiB/s, done.
Resolving deltas: 100% (4457/4457), done.
Note: checking out '79ebb55dcefcfc5d7580f8a2aeb8da6a66461dff'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

Setup.sh: Building rpclib with libc++.
-- The C compiler identification is Clang 8.0.1
-- The CXX compiler identification is Clang 8.0.1
-- Check for working C compiler: /usr/bin/clang-8
-- Check for working C compiler: /usr/bin/clang-8 -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/clang++-8
-- Check for working CXX compiler: /usr/bin/clang++-8 -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test UNUSED_LAMBDA_CAPTURE_WARN_SUPPORTED
-- Performing Test UNUSED_LAMBDA_CAPTURE_WARN_SUPPORTED - Failed
-- Performing Test ZERO_AS_NULL_POINTER_CONSTANT_WARN_SUPPORTED
-- Performing Test ZERO_AS_NULL_POINTER_CONSTANT_WARN_SUPPORTED - Failed
-- Configuring done
-- Generating done
-- Build files have been written to: /home/airl/carla/Build/rpclib-v2.2.1_c3-c8-libcxx-build
[1/14] Building CXX object CMakeFiles/rpc.dir/lib/rpc/this_server.cc.o
clang: warning: -Wl,-L/home/airl/carla/Build/llvm-8.0-install/lib: 'linker' input unused [-Wunused-command-line-argument]
[2/14] Building CXX object CMakeFiles/rpc.dir/lib/rpc/this_session.cc.o
FAILED: CMakeFiles/rpc.dir/lib/rpc/this_session.cc.o 
/usr/bin/clang++-8  -DASIO_STANDALONE -DRPCLIB_ASIO=clmdep_asio -DRPCLIB_FMT=clmdep_fmt -DRPCLIB_MSGPACK=clmdep_msgpack -I/home/airl/carla/Build/rpclib-v2.2.1_c3-c8-source/include -isystem /home/airl/carla/Build/rpclib-v2.2.1_c3-c8-source/dependencies/include -fPIC -std=c++14 -stdlib=libc++ -I/home/airl/carla/Build/llvm-8.0-install/include/c++/v1 -Wl,-L/home/airl/carla/Build/llvm-8.0-install/lib -DBOOST_NO_EXCEPTIONS -DASIO_NO_EXCEPTIONS   -Wall -pedantic -Weverything -Wno-c++98-compat -Wno-c++98-compat-pedantic -Wno-padded -Wno-missing-prototypes -Wno-undef -pthread -std=gnu++11 -MD -MT CMakeFiles/rpc.dir/lib/rpc/this_session.cc.o -MF CMakeFiles/rpc.dir/lib/rpc/this_session.cc.o.d -o CMakeFiles/rpc.dir/lib/rpc/this_session.cc.o -c /home/airl/carla/Build/rpclib-v2.2.1_c3-c8-source/lib/rpc/this_session.cc
clang: warning: -Wl,-L/home/airl/carla/Build/llvm-8.0-install/lib: 'linker' input unused [-Wunused-command-line-argument]
In file included from /home/airl/carla/Build/rpclib-v2.2.1_c3-c8-source/lib/rpc/this_session.cc:1:
In file included from /home/airl/carla/Build/rpclib-v2.2.1_c3-c8-source/include/rpc/this_session.h:6:
/home/airl/carla/Build/llvm-8.0-install/include/c++/v1/atomic:12:9: warning: macro name is a reserved identifier [-Wreserved-id-macro]
#define _LIBCPP_ATOMIC
        ^
In file included from /home/airl/carla/Build/rpclib-v2.2.1_c3-c8-source/lib/rpc/this_session.cc:1:
In file included from /home/airl/carla/Build/rpclib-v2.2.1_c3-c8-source/include/rpc/this_session.h:6:
In file included from /home/airl/carla/Build/llvm-8.0-install/include/c++/v1/atomic:543:
/home/airl/carla/Build/llvm-8.0-install/include/c++/v1/__config:12:9: warning: macro name is a reserved identifier [-Wreserved-id-macro]
#define _LIBCPP_CONFIG
        ^
In file included from /home/airl/carla/Build/rpclib-v2.2.1_c3-c8-source/lib/rpc/this_session.cc:1:
In file included from /home/airl/carla/Build/rpclib-v2.2.1_c3-c8-source/include/rpc/this_session.h:6:
In file included from /home/airl/carla/Build/llvm-8.0-install/include/c++/v1/atomic:544:
/home/airl/carla/Build/llvm-8.0-install/include/c++/v1/cstddef:12:9: warning: macro name is a reserved identifier [-Wreserved-id-macro]
#define _LIBCPP_CSTDDEF
        ^
In file included from /home/airl/carla/Build/rpclib-v2.2.1_c3-c8-source/lib/rpc/this_session.cc:1:
[... 1166 lines cut ...]
using::uint_least16_t;
     ~~^
/home/airl/carla/Build/llvm-8.0-install/include/c++/v1/cstdint:170:8: error: no member named 'uint_least32_t' in the global namespace
using::uint_least32_t;
     ~~^
/home/airl/carla/Build/llvm-8.0-install/include/c++/v1/cstdint:171:8: error: no member named 'uint_least64_t' in the global namespace
using::uint_least64_t;
     ~~^
/home/airl/carla/Build/llvm-8.0-install/include/c++/v1/cstdint:173:8: error: no member named 'int_fast8_t' in the global namespace
using::int_fast8_t;
     ~~^
/home/airl/carla/Build/llvm-8.0-install/include/c++/v1/cstdint:174:8: error: no member named 'int_fast16_t' in the global namespace
using::int_fast16_t;
     ~~^
/home/airl/carla/Build/llvm-8.0-install/include/c++/v1/cstdint:175:8: error: no member named 'int_fast32_t' in the global namespace
using::int_fast32_t;
     ~~^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
5 warnings and 20 errors generated.
ninja: build stopped: subcommand failed.
Util/BuildTools/Linux.mk:143: recipe for target 'setup' failed
make: *** [setup] Error 1
```

Thank you for your time.

NishanthARao, 2020-12-13:
I figured out the error. I had build Unreal Engine already for AirSim usage, and later installed carla. However, I noticed that the UnrealEngine was build with another version of clang. After removing UnrealEngine and reinstalling it from source, and this time using clang8, there were no issues.

Hope this helps!

closed 2020-12-13 by NishanthARao

hexrefugee, 2022-06-19:
> 

I met the same error, hope your function will work.  Thanks.