carla-simulator/carla #3422: The coordinates of the corner of the parking lot

jujuau, 2020-10-11:
Hi,
Thanks for the great simulator. I have a question.
Is it possible to get the coordinates of the corner of the parking lot?(uv, xyz,...)
For example, the red circle in the image below.

[image]

nagasanjay, 2020-10-11:
yes, you can. refer this [thread](https://github.com/carla-simulator/scenario_runner/issues/40)
After running the server, you will get into spectator mode. Move to your desired location to get the coordinates

jujuau, 2020-10-12:
Thank you for your reply.

For example, does it mean that the location (x, y, z) on the map can be obtained by doing the following?

```
spectator = world.get_spectator()  
camera_bp = world.get_blueprint_library().find('sensor.camera.rgb')  
camera_transform = carla.Transform(carla.Location(x=-10,z=10), carla.Rotation(-45,0,0))  
camera = world.spawn_actor(camera_bp, camera_transform, attach_to=vehicle)  
print(spectator.set_transform(camera.get_transform()))
```
output example
→Transform(Location(x=25.259499, y=-49.834530, z=1.488399), Rotation(pitch=-0.008087, yaw=-87.951607, roll=-0.741791))

In the same way, get the coordinates (x, y, z) of the parking lot.
And the UV coordinates can be calculated from the camera's coordinates and the parking lot's coordinates.
Is this understanding correct?

closed 2020-10-17 by jujuau

PTorr, 2020-10-18:
Hi,
Where is the parking lot?

jujuau, 2020-10-20:
Is it the place shown in the image above?
If so, it's in Town 05.

omerler, 2021-06-27:
Hi!
I'm new at Carla as well. 
How did you extract the parking spot's corners, please?
can you extract the all the parking spots in a given image in the simulator? maybe as a mask? maybe as a mask of parking spots instances?

Thanks A lot!

nagasanjay, 2021-06-29:
> Hi @nagasanjay
> I'm new at Carla as well.
> How did you extract the parking spot's corners, please?
> can you extract the all the parking spots in a given image in the simulator? maybe as a mask? maybe as a mask of parking spots instances?
> 
> Thanks A lot!

Hi @omerler. I was just giving a suggestion. I didn't do that myself.

MariamEssam, 2021-07-02:
@jujuau can you please advise which town is this?